adafruit/Adafruit_NeoPixel #280: Driving WS2812 does not work with Wemos D1 Mini @160 MHz instead of 80 MHz

marcel-licence, 2021-07-11:
It looks like that the calculation of the timing does not work when setting arduino to 160MHz.

When using 
```
#define CYCLES_800_T0H  (F_CPU / 2500001) // 0.4us
#define CYCLES_800_T1H  (F_CPU / 1250001) // 0.8us
#define CYCLES_800      (F_CPU /  800001) // 1.25us per bit
```
instead of
```
#define CYCLES_800_T0H  (F_CPU / 2500000) // 0.4us
#define CYCLES_800_T1H  (F_CPU / 1250000) // 0.8us
#define CYCLES_800      (F_CPU /  800000) // 1.25us per bit
```
the timing issue is fixed and WS2812 can be driven by the Wemos D1 Mini @ 80 MHz and also @ 160 MHz.

ladyada, 2021-07-11:
sure please submit a PR

marcel-licence, 2021-07-12:
Pull request submitted: https://github.com/adafruit/Adafruit_NeoPixel/pull/282 thank you

ottobonn, 2022-01-22:
@ladyada looks like this one can be closed!