adafruit/Adafruit_NeoPixel #281: Add support for Arduino Portenta H7

ceilingduster, 2021-07-12:
Please add support for Arduino Portenta H7.  Currently, sketches will compile and upload/run however nothing happens.  

Microcontroller | STM32H747XIH6
-- | --
Architecture | STSTM32
Frequency | 480MHz
Flash | 768KB
RAM | 511.35KB
Vendor | Arduino

ladyada, 2021-07-12:
we'd welcome any PR!

Workshopshed, 2021-08-08:
Not even close to being ready for a PR but a proof of concept.
https://github.com/Workshopshed/Adafruit_NeoPixel

I've added a function that sends data from the Portenta via pin D0.

Based off https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/ and https://forum.arduino.cc/t/fast-gpio-toggle-with-portenta-h7-using-hal/682470

Known limitations
Only tested at 800Khz.
Only tested on the M4 core. 
Hardcoded to pin D0

I'm getting some erratic behaviour on my first pixel but I think that's because it needs a level shifter to get it to 5v. All of the subsequent pixels are fine.

Tested with 2x12 NeoPixel rings and a pair of GRBW neopixel buttons.

razvanphp, 2025-04-08:
Can we please revive this? I'm interested for my project and want to contribute back, maybe we can get to PR-level-quality, whatever that mean.

@jakebierman fixed some of the limitations in his fork:
- https://github.com/Workshopshed/Adafruit_NeoPixel/pull/1
- https://github.com/jakebierman/Adafruit_NeoPixel/

ladyada, 2025-05-04:
this was merged