adafruit/Adafruit_NeoPixel #290: adrafuit class lib can not see code in outside files

karadevnet, 2021-09-15:
ok, i use this cool lib very good in my new project with RPI PICO2040 standard pcb design. Everything works fine, except that i can not make class functions outside from *.ino file. try to make all tricks in c/c++ that i know through the years but have not success. i want to separate functions is separated files and give in INO file standard #include ***.c ***.h but have no success. using same folder not working, include <adafruit_lib> global *.h file not working. Can not find the way to separate all big source code in some *.c *.h files like a normal C/C++ programming. i use arduino ide with rpi pico 2040 board. i try to use extern definition also not success.
https://www.youtube.com/watch?v=JqryNCBsbxU it is link from tube with my test software, but now i have a matrix 20x12 pixel rgb strip and source code growing like a tsunami. file getting bigger and bigger and hard to find where is current code that i test. can sim one to help me how to do it in separated files. When try to make separation says that have not find definitions of functions and variables and all others class members. if have any question that will help for solution i can send to any one who want to help and/or my source code also included

caternuson, 2021-09-15:
It is unclear what the issue is here. Can you provide an small example sketch that demonstrates the issue.

karadevnet, 2021-09-15:
ok, no problemo :)
//-------------------------------------
#include <Adafruit_NeoPixel.h>
#include "colors.h"
#include "effects.h"
#include "effects.c" << when i cut and past some function code from ino file into effects.c compiler says to me can not found functions. i edit the one of files *.ino directly from examples in lib folder

FILE *.ino
//==================
void setup()
{ code works here ; 
} // end setup

void loop()
{  rainbow(100); // when function is in some other file compiler cannot found it
code works and here ;
colorWipe(red, 100);
} // end loop

void colorWipe(uint32_t c, uint8_t wait) {
  for(uint16_t i=0; i<strip.numPixels(); i++) {
    strip.setPixelColor(i, c);
    strip.show();
    delay(wait);
  }
}
end FILE *.ino
//==================

FILE effects.c
//==================

void rainbow(uint8_t wait)
 {  uint16_t i, j;
      for(j=0; j<256; j++) {
    for(i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i+j) & 255));
    }
    strip.show();
    delay(wait);
  }
} // ================
// end FILE effects.c

caternuson, 2021-09-15:
It looks like you are trying to spread your sketch code into multiple files. You are required to have the main file named the same as the sketch folder:
```sketch_name/sketch_name.ino```
and that is required to have `setup()` and `loop()`.

To add other files in your sketch, use Sketch->Add Files... They should show up as new tabs.

Closing since this is not a code issue with this library repo.

closed 2021-09-15 by caternuson

karadevnet, 2021-09-16:
asdasdas

karadevnet, 2021-09-16:
here and all project to find a way, error say that STRIP is not a union or class member or what else is in INO file and can not compile