adafruit/Adafruit_NeoPixel #292: still have error, NOT IN CODE in how it is used ????

karadevnet, 2021-09-16:
ok, i try today to use Sketch->Add Files, and make i files but error code show like this
F:\ELECTRONICS_2021_JUNE\PIXEL_RGB\ADAFRUT_RAINBOW\strandtest_wheel\effects.c: In function 'colorwipe_color':
effects.c:9:16: error: 'strip' undeclared (first use in this function)
for(k=0; k < strip.numPixels(); k++) {
^~~~~
F:\ELECTRONICS_2021_JUNE\PIXEL_RGB\ADAFRUT_RAINBOW\strandtest_wheel\effects.c:9:16: note: each undeclared identifier is reported only once for each function it appears in
Използва се библиотеката „Adafruit_NeoPixel-1.8.5“ от версия 1.8.5 от папка: C:\Users\batkoko\Documents\Arduino\libraries\Adafruit_NeoPixel-1.8.5
exit status 1
Грешка на ред: -1
'strip' undeclared (first use in this function)

i export source code of one function in external file to try to make what you say but have this error
strandtest_wheel.zip
[strandtest_wheel.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/7176095/strandtest_wheel.zip)

here and all project to find a way, error say that STRIP is not a union or class member or what else is in INO file and can not compile
[strandtest_wheel.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/7176106/strandtest_wheel.zip)
if you try to compile in you will see and i am sure you find a way to solve this :)

patlkli, 2021-09-16:
There's lots of issues with your code.

- `effects.c` should be renamed to `effects.cpp`
- Remove `#include "effects.c"` from the main sketch file
- To fix your undeclared identifier problem, either:
  - declare `extern Adafruit_NeoPixel strip;` in your `effects.cpp`, or 
  - pass in `strip` as a reference argument like this:
    ```cpp
    void colorwipe_color(Adafruit_NeoPixel& strip, unsigned int temp_color, unsigned char wait)
    ```

karadevnet, 2021-09-16:
start to edit and hope working, will post answer after test :) 10x very mach, i see need to make some new knowledge for programming :) will try my best

karadevnet, 2021-09-16:
ok, everything is super cool :) . Thanks so very much, I learn some new about programming how to do it because until now I use microchip pic processors and there is little different, but never mind :) working fine super cool and continue to make a matrix effects. Will make a video for end result in tube :) casw closed :)

closed 2021-09-16 by caternuson