adafruit/Adafruit_NeoPixel #295: ESP32 10Mhz cpu frequency NeoPixel goes White and full brightness

peter3099, 2021-09-27:
- Arduino board:  ESP32

- Arduino IDE version (found in Arduino -> About Arduino menu):  ESpressIf 3.3.2

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Hello 
I'm using NeoPixel library to drive a single Sk6812, in my case I'm using IN-PI55QATPRPGPBPW which is a clone that you can get in digikey.

`#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
 #include <avr/power.h> // Required for 16 MHz Adafruit Trinket
#endif

// Which pin on the Arduino is connected to the NeoPixels?
#define PIN        6 // On Trinket or Gemma, suggest changing this to 1

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS 16 // Popular NeoPixel ring size

// When setting up the NeoPixel library, we tell it how many pixels,
// and which pin to use to send signals. Note that for older NeoPixel
// strips you might need to change the third parameter -- see the
// strandtest example for more information on possible values.
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#define DELAYVAL 500 // Time (in milliseconds) to pause between pixels

void setup() {
  // These lines are specifically to support the Adafruit Trinket 5V 16 MHz.
  // Any other board, you can remove this part (but no harm leaving it):
#if defined(__AVR_ATtiny85__) && (F_CPU == 16000000)
  clock_prescale_set(clock_div_1);
#endif
  // END of Trinket-specific code.

  pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
}

void loop() {
  pixels.clear(); // Set all pixel colors to 'off'

  // The first NeoPixel in a strand is #0, second is 1, all the way up
  // to the count of pixels minus one.
  for(int i=0; i<NUMPIXELS; i++) { // For each pixel...

    // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
    // Here we're using a moderately bright green color:
    pixels.setPixelColor(i, pixels.Color(0, 150, 0));

    pixels.show();   // Send the updated pixel colors to the hardware.

    delay(DELAYVAL); // Pause before next pass through loop
  }
}`

This code runs perfectly fine and I can achieve any colour/brightness but in order to reduce power consumption on the ESP32 I need to reduce the cpu frequency using setCpuFrequencyMhz(CPU_MHZ);

The ESP32 runs well on 10,20 or 40MHz but If I put any of those  then when I call pixels.show() I get an unresponsive full brightness full white LED, if I put 8,16 or 24MHz then it runs well but since they are not frequencies that the ESP32 uses then my power consumption is quite high defeating the whole purpose of the CPU frequency reduction. Adafruit NeoPixel library only start working when I setCpuFrequencyMhz(80); above 80MHz

Is there any way to adjust the timing on the library so I can implement the 10MHz CPU frequency for the ESP32?

Thanks in advance
Have a good one

idea--list, 2022-04-22:
In case of the Adafruit ESP32-S3 Feather board (5323) i face the same issue with any CPU frequencies below 80MHz.
Using ESP Arduino core V2.0.3 Dev, Arduino IDE V1.8.19, Adafruit NeoPixel V1.10.4 on a Win10 X64 computer.
Using FastLed library i do not face the issue, so it should be related to Adafruit's implementatioion.

smartua7, 2022-09-02:
Hello! The same problem arose at 12 to 40 and 80 mgz. Does anyone have a solution to this issue in this library?