adafruit/Adafruit_NeoPixel #304: Arduino Nano RP2040 Connect sends signal for different led strips with same pin

HSVDO, 2021-12-07:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **Arduino Nano RP2040 Connect (with/without headers)**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.16**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

When connecting more than 1 led strip to different digital pins the signal strength is added together and sent as output to the first specified pin. The second pin does not output any signal. When I change the order of the specified pins, the summed signal goes to the second (now configured as first) pin. This gives weird colors as result (see 'current behavior.png'). I tested the code with an Arduino Nano Every and it works fine with that kind of board (see 'expected behavior.png'). I can reproduce this problem with 3 different Arduino Nano RP2040 Connect boards following these steps:

[multipleLedStripsExample.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/7671765/multipleLedStripsExample.zip)

1. Connect two neopixel strips to power supply
2. Upload sketch to board
3. Connect the two led strips to the specified pins (e.g. D6 & D7)
4. Connect board to power supply
5. Apply power

OR

1. Upload sketch to board
2. Measure voltage output on specified pins (e.g. D6 & D7)

Current behavior:
[image]

Expected behavior:
[image]

Bodmer, 2021-12-17:
I have posted a pull request #306 that should resolve this. The RP2040 code uses the PIO and had the state machine hard coded, so only one instace was possible.

 The pull request change searches for a free state machine and thus could support up to 8 Neopixel strings (8 state machines availble).

Bodmer, 2021-12-17:
I have created a pull request #307 for a change to the library that permits up to 8 library instances may be used. So up to 8 independant pixel strands can be controlled, each strand on it's own pin.

PaintYourDragon, 2021-12-20:
Issue resolved with PR #307. Thank you!

closed 2021-12-20 by PaintYourDragon

HSVDO, 2021-12-21:
@Bodmer thanks for working on this!
Since I couldn't test it before, I tested the new version 1.10.2, but now I run into another error when compiling:

`ResolveLibrary(Adafruit_NeoPixel.h)
  -> candidates: [Adafruit_NeoPixel@1.10.2]`

`Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:397:3:` error: 'PIO' does not name a type; did you mean 'EIO'?
   PIO pio = pio0;
   ^~~
   EIO`

Using board: 
`BN: Arduino Nano RP2040 Connect
VID: 2341
PID: 005e`

Bodmer, 2021-12-21:
OK, it works fine with Earl Philhower's RP2040 board package but I see that the Arduino package throws the error.

I have created a pull request #308 that fixes this.

The fix is that  the RP2040 #include files need to be in the header file.

HSVDO, 2021-12-21:
It now works fine with the Arduino Nano RP2040 Connect, too. Thanks! :)