adafruit/Adafruit_NeoPixel #305: error while compiling Adafruit_NeoPixel with Attiny85

tadabrowski, 2021-12-08:
Hi!

I'm new to the library. I'm trying any of the example projects using Attiny85 core, but everytime I get a following errors.

```
D:\temp\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp:3392:6: error: prototype for 'void Adafruit_NeoPixel::rainbow(uint16_t, int8_t, uint8_t, uint8_t, bool)' does not match any in class 'Adafruit_NeoPixel'
 void Adafruit_NeoPixel::rainbow(uint16_t first_hue, int8_t reps,
      ^~~~~~~~~~~~~~~~~
In file included from D:\temp\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp:46:0:
D:\temp\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.h:363:8: error: candidate is: void Adafruit_NeoPixel::rainbow(uint16_t, int8_t, uint8_t, uint8_t, boolean)
   void rainbow(uint16_t first_hue = 0, int8_t reps = 1,
        ^~~~~~~
exit status 1
```

- Arduino board:  Attiny85 with drazzy core

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.16

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
1. open any example project
2. compile

caternuson, 2021-12-08:
Same as this:
https://github.com/adafruit/Adafruit_NeoPixel/issues/302

It's been fixed by this PR:
https://github.com/adafruit/Adafruit_NeoPixel/pull/303

Just need to make a new library release, so keep an eye out for that. Anything after the current 1.10.1 version.

ladyada, 2021-12-09:
@dherrada plz bump :)

tadabrowski, 2021-12-09:
@caternuson  thx

closed 2022-01-31 by ladyada