adafruit/Adafruit_NeoPixel #310: Feature

Julius971, 2022-01-10:
Hey, just a minor feature request.
A function like pixels.show(int index);
-  The library should send an update signal for leds 0 to index.  -
This would open a new way for skripting with large amounts of leds. For example, if you want to update led 1 and keep leds 2-1000 unchanged, the transmission would be completed almost instantly compared to the whole 1000+ led update transmission.

ladyada, 2022-01-11:
that is not possible, the entire strip is written at once :)

closed 2022-01-11 by ladyada

Julius971, 2022-01-11:
> that is not possible, the entire strip is written at once :)

But isn't the data sent out bytewise and a led grabs the first 3 bytes and then forwards the rest. 
If you simply truncate the bytes for led n to 1000. the transmission would stop at led n-1