adafruit/Adafruit_NeoPixel #311: Arduino Due and WS2811

danooooo11, 2022-01-18:
Hi, I want to run simple project using Arduino Due and WS2811 LED strip 12V (1 segment = 3 LED) and im just not able to make it. No example program works for me, even if compilation ends with no error, no LED ever light. I just want to verify the compatibility Adafruit_Neopixel vs Arduino Due, becase in forums i cant find any reliable answers. So is really Arduino Due not compatible with your library? Beause if so, i have to buy Arduino Mega and continue in work with it. Many Thanks, Daniel

Board : Arduino Due (original) 
Adafruit_Neopixel Release: 1.10.3
Arduino IDE version: 1.8.19

ottobonn, 2022-01-22:
It looks like the Due (using the ATSAM3X8E) is explicitly supported, so something else might be going wrong.

Here is the code that runs for the Due:

https://github.com/adafruit/Adafruit_NeoPixel/blob/b60a439aee6b9482a778e20e33faeddf01d573bb/Adafruit_NeoPixel.cpp#L2848

ottobonn, 2022-01-22:
You mentioned that you are running the strip at 12V, but note that the Due uses 3.3V logic levels. You might try reducing the supply voltage to the LEDs to see if the problem is that the logic levels are too low. See e.g. this part of the guide: https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels#driving-5v-neopixels-from-3-dot-3v-microcontrollers-2894497-24

ottobonn, 2022-01-22:
See also https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/, linked from other issues here

danooooo11, 2022-01-23:
Hi ottobonn,
yes, I am using Pin 6 (PWM) for sending data to LED Strip DIN and yes, circuit has a common ground. Just like you said, it should work, but it doesnt. 

Here is my super simple example :) maybe i have got some error there and i dont see it :|

`#include <Adafruit_NeoPixel.h>
#define PIN                6 // PWM pin
#define NUMPIXELS   7 // 7 segments of 3 LED, so 21 LED in total
Adafruit_NeoPixel pixels = Adafruit_NeoPixel (NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  pixels.begin();
}

void loop() {
  pixels.setPixelColor(2, pixels.Color(200, 0, 0));
  pixels.show();
  delay(1000);
}`