adafruit/Adafruit_NeoPixel #317: Request to Clarify PIO and SM usage on RP2040

AKLitman, 2022-03-13:
I am using a Cytron Maker PI RP2040 which has two (2) onboard WS2812 LED's.  This library works flawlessly with it.  However, I am also using two other libraries which make use of a PIO and SM (state machines) for Hardware Servo usage and timing an ultrasonic detector based on the US-100.  Can you clarify which PIO and SM this library needs?  And, is there any consideration given to allowing a specific PIO and specific SM to be used?  The other libraries had conflicts but I was to get all three (3) to coexist by declaring usage in a specific order and specific SM's (where allowed).  As PIO usage becomes more popular, I suspect knowing (and defining) specific PIO's and SM's is going to become more prevalent.  Thank you - Austin

The other libraries I am using are:
RP2040_ISR_Servo (by khoih-prog)
NanoConnectHcSr04 (by MrYsLab)

ladyada, 2022-03-13:
what is the specific/exact failure/error you are seeing

AKLitman, 2022-03-13:
I am not seeing a failure of the NeoPixel library.  It works exactly as intended.  My posting was a request on how the NeoPixel library assigns which PIO and SM for use.  Does the NeoPixel library use the PICO SDK functions to request an idle PIO and SM or does the NeoPixel library have a hard coded PIO and SM?

ladyada, 2022-03-13:
dont know but perhaps this code will help!
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L403
https://github.com/adafruit/Adafruit_NeoPixel/blob/abdf456d5690851e748e2e9f91d0d23256651cf5/Adafruit_NeoPixel.cpp#L85

AKLitman, 2022-03-13:
Yes, that helps.  It appears the NeoPixel library has implemented it's own way of finding a free PIO & SM.   I am not literate enough to know if this is how the SDK recommends finding a free PIO & SM, but it works and you have answered my query.  Thank you very much.

closed 2022-03-13 by ladyada