adafruit/Adafruit_NeoPixel #319: Library porting for Arduino Nicla Sense ME

sapienshawk, 2022-03-28:
- Arduino board:  **Arduino Nicla Sense ME*

- Arduino IDE version (found in Arduino -> About Arduino menu):  **ARDUINO IDE 1.8.10 - Windows 10**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **Any standard examples of libary**

Hereafter the complete error message I get when try to compile Neopixel Library with Nicla Sense ME
Arduino: 1.8.19 (Windows 10), Board: "Nicla Sense ME"

C:\Users\sapie\OneDrive\Documenti\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':

C:\Users\sapie\OneDrive\Documenti\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp:2134:24: error: 'g_ADigitalPinMap' was not declared in this scope

pwm->PSEL.OUT[0] = g_ADigitalPinMap[pin];

^~~~~~~~~~~~~~~~

C:\Users\sapie\OneDrive\Documenti\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp:2187:24: error: 'digitalPinToBitMask' was not declared in this scope

uint32_t pinMask = digitalPinToBitMask(pin);

^~~~~~~~~~~~~~~~~~~

C:\Users\sapie\OneDrive\Documenti\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp:2187:24: note: suggested alternative: 'digitalPinToPinName'

uint32_t pinMask = digitalPinToBitMask(pin);

^~~~~~~~~~~~~~~~~~~

digitalPinToPinName

exit status 1

Error compiling for board Nicla Sense ME.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

============================================

[mikeysklar](https://forums.adafruit.com/memberlist.php?mode=viewprofile&u=337713) on Mon Mar 28, 2022 5:41 pm

The Nicla Sense is a nRF52832 (M4 core) board. It looks quite similar to the myNewt that Adafruit has been selling.

You can open an issue requesting support through the NeoPixel github library repo. It might be a matter of adding a board identifier and possibly some pin assignments. The NRF52 / NRF52840 board identifier is already in place in the library.

===============================
So many thanks for any help..
Regards

ladyada, 2022-03-28:
we don't own this board, so we are happy to take a PR but we would not be adding it ourselves since we cannot test

sapienshawk, 2022-03-29:
So many thank for you reply Ladyada (I'm your fan and I really appreciate your work!).
I have the board and I can test the realease if you want...
Mikeskylar from adafruit write: "It might be a matter of adding a board identifier and possibly some pin assignments. The NRF52 / NRF52840 board identifier is already in place in the library."
So it seems no so difficult...
Otherwise I can try by myself if you - please - give me some hints on how to proceed...
Anyway thanks so much!

ladyada, 2022-03-29:
you can look at other PR's for ideas on how to add it, or ask the maker of the board to help add support. it may not be hard but we do not own the board so we can't test.

ladyada, 2022-03-29:
@per1234 fyi, maybe someone who has a nicla can contribute a PR? :)

sapienshawk, 2022-03-30:
I really appreciate your help! (what is PR?) ..
Anyway I'd like to contribute.. but do not have necessary know how.
I can if you give me hints... otherwise may I contribute in some way with money?
Thanks so much regards.

sapienshawk, 2022-04-22:
Dear sirs,
no hope to have support on this issue?
Can you please redirect me to some paid service?
Thanks regards

Alexandra182, 2022-10-05:
Hello! I had the same issue so I tried to fix it myself. I've made a PR that fixes this issue, you can find it [here](https://github.com/adafruit/Adafruit_NeoPixel/pull/331).
Let me know if it works for you!

sapienshawk, 2022-10-05:
Hello! I will check as soon as possible! Meanwhile thank so much!!!! Great!
ing.Amedeo Lepore
mail: ***@***.***
phone: +39 348 7017 257

Cali1205, 2022-12-04:
Hi @Alexandra182,

thanks for the fix.
I tried it and I'm facing the issue that somehow the PIN is not set.
So it compiles and upload --> no issue.
But then there is no reaction and also the initial setting of the Output Pin to LOW is not done.

Anybody else tried it out and perhaps can provide some working code?

Here is mine:

```
#include <Arduino.h>
#include "Nicla_System.h"
#include "Arduino_BHY2.h"
#include <Adafruit_NeoPixel.h>

SensorOrientation orientation(SENSOR_ID_ORI);

#define LED_PIN GPIO3
#define LED_COUNT 10
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup(){
  Serial.begin(115200);
  BHY2.begin();
  orientation.begin();
  nicla::begin();
  nicla::leds.begin();
  strip.begin();
}
```

Thanks a lot!

Alexandra182, 2022-12-05:
Hi @Cali1205, 

If you want to connect the NeoPixels to GPIO3 you can use this line of code:

`#define PIN  0`

according to the [pinout](https://content.arduino.cc/assets/ABX00050-pinout.png).

`#define LED_PIN GPIO3` didn't work for me either.

This is the code that worked for me:

```
#include <Adafruit_NeoPixel.h>

#define PIN       0 
#define NUMPIXELS 1 

Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#define DELAYVAL 500

void setup() {
  pixels.begin();
}

void loop() {
  pixels.clear();

  for(int i=0; i<NUMPIXELS; i++) {
    pixels.setPixelColor(i, pixels.Color(random(250), random(250), random(250)));
    pixels.show();
    delay(DELAYVAL);
  }
}
```

Cali1205, 2022-12-05:
Thanks a lot @Alexandra182,

that works fine for me.
It seems that the use of the Pindescription used for the mapping of GPIO3 --> Pin 10 is not working properly...

[image]