adafruit/Adafruit_NeoPixel #323: Starting LED index is not zero. Strip is shifted by 20.

ericcastro, 2022-06-30:
I have this LED string I bought off Aliexpress which are supposed to be SK6812 leds and WS2812B compatible.

https://www.aliexpress.com/item/1005001917228863.html?spm=a2g0o.order_list.0.0.21ef1802b7v2j4

The "strandtest" example code 99% of the time just gets it right for most LED strips I tried, but in this case I encountered something I never seen or thought was possible.

I have 20 LEDs on this string, so I set LED_COUNT to 20.

Upload the sketch. Nothing happens. After trying a million things I randomly gave increasing the LED count a try, and then I saw that numbers above 20 made the string start to light up.

A few more tries and figured I should set LED_COUNT to 40, then all leds lighted up.

At first try I got all the wrong colors, it was because the default was set to RGBW. So I changed the type to NEO_RGB, and I got the right colors.

Obviously, 40 was not my led count but 20, so what I could find out then is that the first pixel is on index 20, then on 21, 22, etc.

Is this even possible, circuit wise ?

I couldn't find anything in the code allowing to shift first-pixel index by a certain amount or anything, so I am a little lost on what can possibly cause this.

caternuson, 2022-06-30:
They probably have slightly different timings than ones this library has been tested on.

closed 2022-06-30 by caternuson

ericcastro, 2023-01-16:
I have noticed this is always the case with these strips but couldn't explain why.

unlike most other strips, each led seems to have its own fixed index.

let's say you have one of these, with leds numbered from 0 to 99

if you cut this strip in half with scissors, you don't get two "0 to 49" strips. instead, you get one that goes 0 to 49, and the second one, goes from 50 to 99, and your total number of leds is still 100 - but you start counting from 50)

again, I could not explain how this is even possible but these different chips may have some sort of memory where the index is stored.

I'm glad I'm able to still use them nonetheless, just shift the index count by whatever offset is needed.