adafruit/Adafruit_NeoPixel #332: ESP32 S2 arduino 2.0.5 crashes?

tablatronix, 2022-10-08:
I upgraded from 2.0.4 to 2.0.5

Stack seems to be here last
```c++
    rmt_config(&config);
    rmt_driver_install(config.channel, 0, 0);
```

```php
Exception Cause: Not found

0x400273ca: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
0x4002c8f1: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x400319ad: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
0x40028851: _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:167
0x400957d2: rmt_driver_install at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/rmt.c:1092
0x4008ca26: espShow at /Users/shawn/.platformio/lib/Adafruit NeoPixel/esp.c:129
0x4008c7d9: Adafruit_NeoPixel::show() at /Users/shawn/.platformio/lib/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:2885
0x40084b0f: init_indicator(unsigned short) at /Users/shawn/projects/microcontrollers/dev/libraries/SHAWN_universal/neoindicator.h:62
0x4008589a: setup() at /Users/shawn/projects/microcontrollers/dev/libraries/SHAWN_boards/esp_env/esp_env.ino:203
0x40092436: loopTask(void*) at /Users/shawn/.platformio/packages/framework-arduinoespressif32@src-f2ea83e2545300b10a69ff44ef9dc6cd/cores/esp32/main.cpp:42
```

schoolpost, 2022-11-11:
Experiencing similar with 2.0.4 and 2.0.5 on a standard ESP32 ( ESP32-DOWD-V3 )

tablatronix, 2022-11-11:
I have not, I will test again and see if it is still present here

schoolpost, 2022-11-12:
> I have not, I will test again and see if it is still present here

Turns out code was crashing for another reason, working on 2.0.5 and latest Neopixel library version.