adafruit/Adafruit_NeoPixel #338: NeoPixel ESP8266 does not compile. not with Arduino 1.8 not with Arduino 2.0.1

Jens869, 2022-11-10:
- Arduino board:  ESP8266/Generic ESP8266

- Arduino IDE version (found in Arduino -> About Arduino menu):   2.0.1 and 1.8.19

-  examples/ Adafruit Neopixel / simple
- or any other sample script

start compiling

...
Linking everything together...
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/mkdir.py -p /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/ld_h/
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/cp.py /Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/ld/eagle.flash.1m64.ld /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/ld_h/local.eagle.flash.ld.h
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/ld_h/local.eagle.flash.ld.h -o /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/local.eagle.flash.ld
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 /Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/ld/eagle.app.v6.common.ld.h -o /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/local.eagle.app.v6.common.ld
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/simple.ino.map -g -Werror=return-type -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lib -L/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lib/NONOSDK22x_190703 -L/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628 -L/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/libc/xtensa-lx106-elf/lib -Tlocal.eagle.flash.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/simple.ino.elf -Wl,--start-group /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/sketch/simple.ino.cpp.o "/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp 2.c.o" /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp.c.o "/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp8266 2.c.o" /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp8266.c.o "/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/kendyte_k210 2.c.o" /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/kendyte_k210.c.o "/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o" /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/core/core.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp8266.c.o: in function `espShow':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266.c:21: multiple definition of `espShow'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/esp8266 2.c.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266 2.c:21: first defined here
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o: in function `_ZN17Adafruit_NeoPixelC2Ev':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:113: multiple definition of `_ZN17Adafruit_NeoPixelC2Ev'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:113: first defined here
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o: in function `_ZN17Adafruit_NeoPixelC2Ev':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:113: multiple definition of `_ZN17Adafruit_NeoPixelC1Ev'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:113: first defined here
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o: in function `_ZN17Adafruit_NeoPixelD2Ev':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:119: multiple definition of `_ZN17Adafruit_NeoPixelD2Ev'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:119: first defined here
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o: in function `_ZN17Adafruit_NeoPixelD2Ev':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:119: multiple definition of `_ZN17Adafruit_NeoPixelD1Ev'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:119: first defined here
/Users/jensgerritschroder/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o: in function `_ZN17Adafruit_NeoPixel5beginEv':
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:128: multiple definition of `_ZN17Adafruit_NeoPixel5beginEv'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:128: first defined here
[... cut ...]
/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:3436: multiple definition of `_ZN17Adafruit_NeoPixel7rainbowEtahhb'; /var/folders/gm/4sy77v7s78v5sl_h8jhx7p6w0000gn/T/arduino_build_754628/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp.o:/Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel 2.cpp:3435: first defined here
collect2: error: ld returned 1 exit status
Bibliothek Adafruit_NeoPixel in Version 1.10.6 im Ordner: /Users/jensgerritschroder/Documents/Arduino/libraries/Adafruit_NeoPixel  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

moose4lord, 2022-11-10:
I could not reproduce this issue. Compile and link worked fine for me using IDE v1.8.19 and Adafruit_Neopixel v1.10.6.
Here's my log:
```c++
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/klord/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/klord/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/klord/Desktop/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_build_584095 -warnings=none -build-cache /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_cache_924284 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_modified_sketch_163569/simple.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/klord/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/klord/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/klord/Desktop/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0000_0000 -ide-version=10819 -build-path /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_build_584095 -warnings=none -build-cache /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_cache_924284 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mkspiffs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/mklittlefs/3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/klord/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_modified_sketch_163569/simple.ino
Using board 'generic' from platform in folder: /Users/klord/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Using core 'esp8266' from platform in folder: /Users/klord/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2
Detecting libraries used...
/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -I/Users/klord/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/include -I/Users/klord/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lwip2/include -I/Users/klord/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_build_584095/core -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-excep [... cut ...]
Using library Adafruit_NeoPixel at version 1.10.6 in folder: /Users/klord/Desktop/Arduino/libraries/Adafruit_NeoPixel 
/Users/klord/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-size -A /var/folders/5r/r30866kj6c1gqr371yy69gl40000gn/T/arduino_build_584095/simple.ino.elf
Sketch uses 261837 bytes (27%) of program storage space. Maximum is 958448 bytes.
Global variables use 28020 bytes (34%) of dynamic memory, leaving 53900 bytes for local variables. Maximum is 81920 bytes.
```

Jens869, 2022-11-10:
my solution:
Quit Arduino
rename the directory /Arduino/libraries to /Arduino/libraries-old  (mv )
Start Arduino
Install required Libraries

the libraries were also up to date before, I don't know why it works now.

Jens869, 2022-11-10:
Library directory deleted and created again

closed 2022-11-10 by Jens869