adafruit/Adafruit_NeoPixel #341: Adafruit NeoPixel on Arduino IoT web page issue: can't compile

oli6120, 2022-12-17:
/usr/local/bin/arduino-cli compile --fqbn esp8266:esp8266:d1:baud=921600,dbg=Disabled,eesz=4M,exception=disabled,ip=lm2f,lvl=None____,vt=flash,wipe=none,xtal=80 --libraries /home/builder/opt/libraries/latest --build-cache-path /tmp --output-dir /tmp/490405119/build --build-path /tmp/arduino-build-5BA362754C3A95663B89BCFDAEC7B806 --library /home/builder/opt/libraries/dht_sensor_library_1_4_4 --library /home/builder/opt/libraries/adafruit_neopixel_1_10_6 /tmp/490405119/Bureau_dec16a

Using library arduinomqttclient_0_1_6 at version 0.1.6 in folder: /home/builder/opt/libraries/arduinomqttclient_0_1_6

Using library adafruit_unified_sensor_1_1_6 at version 1.1.6 in folder: /home/builder/opt/libraries/adafruit_unified_sensor_1_1_6

/home/builder/opt/libraries/adafruit_neopixel_1_10_6/esp8266.c:20:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'

IRAM_ATTR void espShow(

^

Error during build: exit status 1

here is my Arduino iot project:

// DHT sensor library - Version: 1.4.4
#include <DHT.h>
#define DHTPIN 2
#define DHTTYPE DHT11
DHT dht(DHTPIN, DHTTYPE);

// Adafruit NeoPixel - Version: 1.10.6
#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
#include <avr/power.h>
#endif
#define PIN            4
#define NUMPIXELS      18
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#include "thingProperties.h"

void setup() {
  // Initialize serial and wait for port to open:
  Serial.begin(9600);
  delay(1500); 

  // Defined in thingProperties.h
  initProperties();
  
  //ledRGB
  pixels.begin();
  
  //init T & H sensor
  dht.begin();

  // Connect to Arduino IoT Cloud
  ArduinoCloud.begin(ArduinoIoTPreferredConnection);
  
  /*
     The following function allows you to obtain more information
     related to the state of network and IoT Cloud connection and errors
     the higher number the more granular information you’ll get.
     The default is 0 (only errors).
     Maximum is 4
 */
  setDebugMessageLevel(2);
  ArduinoCloud.printDebugInfo();
}

void loop() {
  ArduinoCloud.update();
  DHT_SENSOR_READ();
  // Your code here 
  
  
}

/*
  Since Humidity is READ_WRITE variable, onHumidityChange() is
  executed every time a new value is received from IoT Cloud.
*/
void onHumidityChange()  {
  // Add your code here to act upon Humidity change
}

/*
  Since Neopixel is READ_WRITE variable, onNeopixelChange() is
  executed every time a new value is received from IoT Cloud.
*/
void onNeopixelChange()  {
  // Add your code here to act upon Neopixel change
uint8_t r, g, b;
neopixel.getValue().getRGB(r, g, b);
if (neopixel.getSwitch()) {
Serial.println("R:" + String(r) + " G:" + String(g) + " B:" + String(b)); //prints the current R, G, B values
for (int i = 0; i < NUMPIXELS; i++) {
// pixels.Color takes RGB values, from 0,0,0 up to 255,255,255
pixels.setPixelColor(i, pixels.Color(r, g, b));
pixels.show(); // This sends the updated pixel color to the hardware.
}
}
else {
Serial.println("Lamp Off");
//the following code simply turns everything off
for (int i = 0; i < NUMPIXELS; i++) {
// pixels.Color takes RGB values, from 0,0,0 up to 255,255,255
pixels.setPixelColor(i, pixels.Color(0, 0, 0));
pixels.show(); // This sends the updated pixel color to the hardware.
}
}
}

void DHT_SENSOR_READ() {
  float h = dht.readHumidity();
  float t = dht.readTemperature();
  temperature = t;
  humidity = h;
  Serial.print("Température - "); Serial.println(t);
  Serial.print("Humidité - "); Serial.println(h);
  delay(1000);
}

tanks in advance

caternuson, 2022-12-18:
Can't compile the test sketch without `thingProperties.h`. Is this coming from a guide? If so, can you link to the guide where the example originates.

oli6120, 2022-12-20:
I supposed it's credentials and wifi access auto-generated by Arduino IoT cloud
When you get started a new IoT project.
when you get started, You must add an esp8266 or an Arduino product IoT available and then Arduino auto-create a "ThingProperties.h" files with you ssid, PW, esp, etc

link: https://cloud.arduino.cc

here is the file thinProperties.h

// Code generated by Arduino IoT Cloud, DO NOT EDIT.

#include <ArduinoIoTCloud.h>
#include <Arduino_ConnectionHandler.h>

const char DEVICE_LOGIN_NAME[]  = ".....";

const char SSID[]               = SECRET_SSID;    // Network SSID (name)
const char PASS[]               = SECRET_OPTIONAL_PASS;    // Network password (use for WPA, or use as key for WEP)
const char DEVICE_KEY[]  = SECRET_DEVICE_KEY;    // Secret device password

void onNeopixelChange();
void onHumidityChange();

CloudColoredLight neopixel;
CloudTemperatureSensor temperature;
CloudRelativeHumidity humidity;

void initProperties(){

  ArduinoCloud.setBoardId(DEVICE_LOGIN_NAME);
  ArduinoCloud.setSecretDeviceKey(DEVICE_KEY);
  ArduinoCloud.addProperty(neopixel, READWRITE, ON_CHANGE, onNeopixelChange);
  ArduinoCloud.addProperty(temperature, READ, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(humidity, READWRITE, ON_CHANGE, onHumidityChange);

}

WiFiConnectionHandler ArduinoIoTPreferredConnection(SSID, PASS);

simsclub, 2023-01-12:
Seems that ESP8266 boards fail to compile after v1.8.4 of this library. Tested using https://create.arduino.cc/ with Generic ESP8266 and NodeMCU 1.0 for the board which both work with v1.8.4 but nothing I tried after. 

Compiling on ESP32-CAM board for instance, compiles ok using Adafruit_NeoPixel.h v1.10.7.

simsclub, 2023-01-13:
I was testing with https://create.arduino.cc/ as mentioned, and because oli6120 referred to that IDE in their message of 20/12/2022.

If you mean `strandtest.ino`, that doesn't build in the above IDE using `Adafruit_NeoPixel.h` library >= v1.8.5

However, `strandtest.ino` doesn't build using v1.8.4 until you comment out the `strip.rainbow()` because I presume, it was added to `Adafruit_NeoPixel.h` after v1.8.4. Once you comment out the `strip.rainbow()` call, the `strandtest.ino` code builds without error.

per1234, 2024-01-16:
Here is a minimal demonstration sketch to reproduce the fault:

```cpp
#include <Adafruit_NeoPixel.h>
void setup() {}
void loop() {}
```

There is an incompatibility between the "Adafruit NeoPixel" library and version 2.5.2 and older of the ["esp8266" boards platform](https://github.com/esp8266/arduino). I bisected the regression to fee9f57230a9a99ba6b5046b2b812d30c84c0dc8.

## Workarounds

### Arduino IDE

For Arduino IDE users, the best workaround/solution will be to use Boards Manager to update to the modern version of the "esp8266" boards platform. You can follow the official instructions here:

https://github.com/esp8266/arduino#installing-with-boards-manager

### Arduino Cloud

[**Arduino Cloud**](https://cloud.arduino.cc/) uses version 2.5.0 of the "esp8266" boards platform. There is no way for the user to update to a compatible version of the platform so the easiest workaround will be to configure your sketch to use the last version of the "Adafruit NeoPixel" library from before the backwards incompatibility was introduced:

1. Open your sketch in "**Arduino Web Editor**".
1. From the menu on the left side of the "**Arduino Web Editor**" window, click "**Libraries**".
1. Click the "**LIBRARY MANAGER**" button.
   The "**Favorite the Contributed libraries you use the most**" dialog will open.
1. Type `Adafruit NeoPixel` in the "**SEARCH LIBRARY**" field.
1. Find the "**Adafruit NeoPixel**" library in the search results.
1. If the star to the right of the library is not already filled, click the star.
   This will add the library to your "**Favorites**" in "**Arduino Web Editor**".
1. Click the "**DONE**" button.
   The "**Favorite the Contributed libraries you use the most**" dialog will close.
1. From the **Libraries** panel of the "**Arduino Web Editor**" window, click the "**FAVORITES**" tab.
1. Find the "**Adafruit NeoPixel**" library in the list of favorite libraries and hover the mouse pointer over its name.
   An "**INCLUDE ▼**" button will appear to the right side of the library name.
1. Click the **▼** on the right side of the "**INCLUDE ▼**" button.
   A menu will open.
1. Select "**1.8.4**" from the menu.

This configures the metadata of the currently opened sketch so that Arduino Cloud will always use version 1.8.4 of the "Adafruit NeoPixel" library when the sketch is compiled. The configuration is per-sketch, so you will need to repeat the above procedure for each sketch using the "Adafruit NeoPixel" library that you want to use with an ESP8266 board.