adafruit/Adafruit_NeoPixel #349: Requesting Arduino Giga Wifi R1 Support (STM32H747)

MoveElectricMobility, 2023-03-19:
Hello everyone! I recently moved from an Arduino Uno Wifi Rev 2 to the Arduino Giga Wifi R1 (STM32H747 based MCU) and this library stopped working, likely because support beyond the older Due board has not been built yet. If someone could add support for the Giga I would be very grateful, the new board is extremely popular. Thanks!

donemuhendislik, 2023-03-24:
Sure, we need an urgent support for this topic too... All other libraries as FastLED is uncompatible with Arduino Giga Wifi R1 too... Thanks for your helps

Haschtl, 2023-07-18:
I've created a fork which works with the Arduino GIGA. You can check it out [here](https://github.com/Haschtl/Adafruit_NeoPixel/tree/master).

It may not be the optimal implementation, but it works.

Thanks to the takuya ichise for the codebase for STM32 ([here](https://os.mbed.com/users/tichise/code/NeoPixel/)).

Here is the main part (`show()`) of the implementation I added:
```
#if defined(TARGET_GIGA)
  // Arduino GIGA -----------------------------------------------------------
  uint8_t *p = pixels, *end = p + numBytes, pix;
  while (p < end)
  {
    pix = *p++;
    for (int i = 0; i < 8; i++)
    {
      // gpio_write(&gpio->gpio, 1);
      gpio->write(1);

      // duty cycle determines bit value
      // if (pix & 0x80)
      if (bitRead(pix, i) == 0)
      {
        // one
        // wait_ns(400); -> 192 cycles
        for (int j = 0; j < 96; j++)
          __NOP();

        // gpio_write(&gpio->gpio, 0);
        gpio->write(0);

        // wait_ns(850) -> 408 cycles
        for (int j = 0; j < 204; j++)
          __NOP();
      }
      else
      {
        // zero
        // wait_ns(800) -> 384 cycles
        for (int j = 0; j < 192; j++)
          __NOP();

        gpio->write(0);
        // gpio_write(&gpio->gpio, 0);
        // wait_ns(450) -> 216 cycles
        for (int j = 0; j < 108; j++)
          __NOP();
      }

      // pix = pix << 1; // shift to next bit
    }
  }
```

RobUnplanned, 2023-08-13:
I have been used your fork - I am happy that it does give signal to the NeoPixel - but when doing a rainbow effect to test the functionality - noticed that it has trouble with finer RGBW values - so unsure if I have a different set up to you, or something else? I am using a Giga, and a typical hue loop.

tineira, 2023-11-16:
@Haschtl thanks! its working perfectly.

Haschtl, 2023-11-16:
> I have been used your fork - I am happy that it does give signal to the NeoPixel - but when doing a rainbow effect to test the functionality - noticed that it has trouble with finer RGBW values - so [...]

It depends on the library you are using for the NeoPixel. At the time I was programming my Giga project, I couldn't find any officially supporting the Giga. The timing in these libraries needs to be adapted for the Giga. It's also different for the main core and the second core.

My final solution that I can suggest when using NeoPixel is using the second core to control the NeoPixel pins and control it with via shared memory from the main core. This way your main-core is free and you have no problems with interrupts, because the timing critical part runs entirely on another core.

RubiCubix, 2024-03-07:
Nice work. I noticed two things while I tested the fork.  I am using ws2812b rgb led strips. 

1. For me it seems like the RGB order has changed compared to running the same code on my Arduino MEGA using the original Adafruit neopixel library. 

2. I am unable to set the color of the first neopixel on each led strip (separate IO pins). It's seems like when show() is called the first pixel i set to some random color no matter what. For instance, when I clear all pixels the first pixel remains green in my test.

Anyway, thanks for coming up with this code. It's one step towards stable support for the GIGA R1.

ABCartHub, 2024-03-12:
Hello everyone,
Thank you @Haschtl for your fork to adapt the library to GIGA R1! It's really useful to me! Great job.
Like @RubiCubix, the first pixel of my strip lights green when all should be off. Did someone found a workaround?
Thanks in advance!

j-bober, 2024-04-10:
Hi guys, to solve the problem @RubiCubix and @ABCartHub have, change the following line from @Haschtl 's original post:

_**Original line:**_
uint8_t *p = pixels, *end = p + numBytes, pix;

_**Updated line:**_
uint8_t *p = pixels, *end = p + numBytes-1, pix;

This solved the issue for me.

geerbel, 2024-04-23:
First of all, thanks to @Haschtl for providing a fork

As stated by @ABCartHub and @RubiCubix , I had the same issue that the first pixel always remained green. In addition, also the brightness settings messed up the correct color (brightness was not really working as a setting).

As reference, below the used components:
- Arduino Giga with the display shield
- a grove mega shield 
- the grove 10 led stick which is based on the WS2813 mini (https://www.seeedstudio.com/Grove-RGB-LED-Stick-10-WS2813-Mini.html)
- The code is targetting the main cortex M7

However I seem to found a solution, which works in my case. 
In the "Adafruit_NeoPixel.cpp" code of @Haschtl ; I changed the value of 'j' in the following code lines.
- The modified code starts at line 1692 and ends on line 1728
- I have marked the changed values in bold

`#if defined(TARGET_GIGA)
        for (int j = 0; j < **80**; j++)
#else
        for (int j = 0; j < **40**; j++)
#endif
          __NOP();

        // gpio_write(&gpio->gpio, 0);
        gpio->write(0);

        // wait_ns(850) -> 408 cycles
#if defined(TARGET_GIGA)
        for (int j = 0; j < **170**; j++)
#else
        for (int j = 0; j < **85**; j++)
#endif
          __NOP();
      }
      else
      {
        // zero
        // wait_ns(800) -> 384 cycles
#if defined(TARGET_GIGA)
        for (int j = 0; j < **160**; j++)
#else
        for (int j = 0; j < **80**; j++)
#endif
          __NOP();

        gpio->write(0);
        // gpio_write(&gpio->gpio, 0);
        // wait_ns(450) -> 216 cycles
#if defined(TARGET_GIGA)
        for (int j = 0; j < **90**; j++)
#else
        for (int j = 0; j < **45**; j++)
#endif`

I doubt these are the 'perfect values', however now everything works. 
- I can set each of my 10 pixels to the correct RGB value (tested some colors)
- Setting the brightness functions

Just for future reference in case somebody still needs it, as the current official AdaFruit Neopixel Library is still not supporting the Arduino Giga board.

tineira, 2024-08-24:
I found the problem and a workarround for color correction. for some reason you have to send the colors with the bit order inverted.

you can get the correct color with this functions 
```

uint8_t reveserBits(uint8_t val) {
    uint8_t valRev = 0;
    uint8_t bitMask  = B00000001;
    uint8_t bitWrite = B10000000;

    for(int i = 0; i < 8; i++) {
      if ((val & bitMask) > 0) {
        valRev = valRev | bitWrite;
      }
      bitMask = bitMask << 1;
      bitWrite = bitWrite >> 1;
    }

    return valRev;
}

 uint32_t getColor(uint8_t r, uint8_t g, uint8_t b){
     uint32_t color = reveserBits(r);
     color = color << 8;
     color += reveserBits(g);
     color = color << 8;
     color += reveserBits(b);

     return color;
 }
```

tineira, 2024-08-24:
> I've created a fork which works with the Arduino GIGA. You can check it out [here](https://github.com/Haschtl/Adafruit_NeoPixel/tree/master).
>
> It may not be the optimal implementation, but it [...]

if you change the for loop for this code, the correct colors are shown
`for (int i = 7; i >= 0; i--)`

RubiCubix, 2025-02-19:
I spent some time trying to get it working with FastLED and it seems like I actually got something running. 
The basic Blink and Blur example projects in FastLED works alright on a WS2812b led ring with 45 leds and signal wire connected to D3 on the Arduino GIGA R1 Board.  I have not done any more testing yet. 

Feel free to try it out
https://github.com/RubiCubix/FastLED/tree/Arduino-GIGA-R1-WiFi-Support

tyeth, 2025-05-09:
Please checkout the latest version, released a few minutes ago. I believe it may add the GIGA support desired in this issue (so maybe it can be closed).