adafruit/Adafruit_NeoPixel #358: ESP32 light WS2812B RED color BUG

blackberryer, 2023-06-12:
ESP32 lights a single WS2812B (side-illuminated version), using the reference routine 'RGBWstrandtest.ino', because it is a single piece, modify the following variables:
`#define LED_COUNT  1`
I didn't find any problems when I lit 'BLUE' and 'GREEN';
But changing the color parameter to 'RED' has a problem:
Light up the blue light and green light separately, and display the blue light/green light normally;
Light the red light alone, it flashes constantly, and you can see a short blue light;

```
void loop() {
  // Fill along the length of the strip in various colors...
  colorWipe(strip.Color(255,   0,   0)     , 50); // Red
  //colorWipe(strip.Color(  0, 255,   0)     , 50); // Green
  //colorWipe(strip.Color(  0,   0, 255)     , 50); // Blue
  //colorWipe(strip.Color(  0,   0,   0, 255), 50); // True white (not RGB white)

  //whiteOverRainbow(75, 5);

  //pulseWhite(5);

  //rainbowFade2White(3, 3, 1);
}
```

caternuson, 2023-06-12:
If you change the loop to this:
```cpp
void loop() {
  strip.fill(strip.Color(255, 0, 0));
  strip.show();
}
```
what happens?

blackberryer, 2023-06-12:
> If you change the loop to this:
> 
> ```c++
> void loop() {
>   strip.fill(strip.Color(255, 0, 0));
>   strip.show();
> }
> ```
> 
> what happens?

Wow, that's amazing, successfully solved my confusion, thank you so much.

Let me see what happened and what difference they had.

blackberryer, 2023-06-13:
The test again found that when set to red, the delay time must be less than 5, otherwise flickering will occur.

```
void colorWipe(uint32_t color, int wait) {
  for(int i=0; i<strip.numPixels(); i++) { // For each pixel in strip...
    strip.setPixelColor(i, color);         //  Set pixel's color (in RAM)
    strip.show();                          //  Update strip to match
    delay(wait);                           //  Pause for a moment
  }
}
```

The value of the above code 'wait' must be less than 5.

blackberryer, 2023-06-13:
The problem was solved, it was a problem with the power supply, and the increase in capacitance or the software reduced the brightness found that it would not flicker, and finally the series protection resistance was reduced to 620R on the power supply and the capacitance was increased by 4.7uf, since the previous APA102 was used, they recommended series resistance

closed 2023-06-13 by blackberryer