adafruit/Adafruit_NeoPixel #363: I'm interested in supporting uno r4 wifi

Lormos, 2023-08-02:
Can you direct me on what should i look first and what are the problems right now?

KurtE, 2023-09-15:
Sorry, I am not an expert on this library, but have played with it some:

What probably needs to happen is to add a new section of code into Adafruit_NeoPixel.cpp, 
Into the monolithic function: void Adafruit_NeoPixel::show(void) 

```
First look after the AVR section for the code that looks like:
```
  // END AVR ----------------------------------------------------------------

#elif defined(__arm__)

```
And then you will see a lot of different types of boards under it:
#if defined(ARDUINO_ARCH_RP2040)
...
#elif defined(TEENSYDUINO) &&                                                  \
    defined(KINETISK) // Teensy 3.0, 3.1, 3.2, 3.5, 3.6
...
```
Maybe to about: 
```
TC_Stop(TC1, 0);

#endif // end Due

  // END ARM ----------------------------------------------------------------

```
And add a new #elif defined(
For eiither specific board or boards or the like.
You will then need to write the code to output the pulses at the right time.  And some of these Neopixels need these pretty precise to work.

Some do it by adding just the right number of NOPs to work.  Others might count instruction cycles, or maybe use a timer.

Sorry I know not much help.

eMUQI, 2023-10-18:
It simply doesn't work for me.

I attempted to utilize the `standtest` example code from the library on my R4WIFI board, but my WS2812 RGB LED strip does not respond(None of the LEDs light up). However, there are no issues with my R3 board.