adafruit/Adafruit_NeoPixel #366: .show() not working on RPi Pico

Xarbenence, 2023-09-21:
Hello,

I am trying to use this library on the Raspberry Pi Pico, built within the Arduino environment within Platform.IO

Does this library not support the pico board?

PaintYourDragon, 2023-09-22:
Tested, works on Pico, Arduino IDE w/Earle Philhower RP2040 core. Unsure about Platform.IO.

euphemism, 2023-09-28:
Also attempting to light up some NeoPixels using an RP2040, the Earle Philhower core, and PlatformIO. Seems to hang when `show()` is called.

PaintYourDragon, 2023-10-09:
I wonder if Platform.IO does something differently with setting up or assigning PIO. Might’ve encountered this before, it sounds almost familiar.

In Adafruit_NeoPixel.h, near the bottom, look for:
`  PIO pio = pio0;`
Change pio0 to pio1 and see if any difference. If so, maybe we put an #ifdef around it for Platform.IO.

taunoe, 2023-10-11:
I got mine to work with PlatformIO
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
and PIO pio = pio0;

ladyada, 2023-10-28:
closing due to no response & not reproducable

closed 2023-10-28 by ladyada