adafruit/Adafruit_NeoPixel #371: 1.12.0 RMT Error on ESP32-C2

rcmontrose, 2023-11-09:
I just updated to the 1.12.0 version of the Library, and when running on an ESP32-C3 board, when running the NeoPixel Ring simple sketch, it doesn't display on the LED strip and gives errors when communicating to the LEDS. The code works fine with the 1.11.0 library, though it gives the deprecated rmt warning.

```
  void setup() {
    Serial.begin(2000000);
    Serial.setDebugOutput(true);
    delay(2000);
  
    Serial.println("Setup");
    pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
    Serial.println("pixels.begin");
  }
  
  void loop() {
    Serial.println("Loop");
    pixels.clear(); // Set all pixel colors to 'off'
    Serial.println("pixels.clear");
  
    // The first NeoPixel in a strand is #0, second is 1, all the way up
    // to the count of pixels minus one.
    for(int i=0; i<NUMPIXELS; i++) { // For each pixel...
  
      // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
      // Here we're using a moderately bright green color:
      pixels.setPixelColor(i, pixels.Color(0, 150, 0));

      pixels.show();   // Send the updated pixel colors to the hardware.
      Serial.println("pixels.show #"+String(i));
  
      delay(DELAYVAL); // Pause before next pass through loop
    }
    Serial.println("Complete");
```

```
  Setup
  pixels.begin
  Loop
  pixels.clear
  E (2135) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
  [  2104][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  2104][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  E (2136) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  2106][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  2107][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #0
  E (2638) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  2607][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  2607][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #1
  E (3139) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  3108][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  3108][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #2
  E (3640) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  3609][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  3609][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #3
  E (4141) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  4110][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  4110][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #4
  E (4642) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1070159236
  [  4611][E][esp32-hal-rmt.c:518] rmtInit(): GPIO 10 - RMT TX Initialization error.
  [  4611][E][esp.c:41] espShow(): Failed to init RMT TX mode on pin 10
  pixels.show #5
...... and so on for all the pixels......
```

Using Arduino IDE 2.2.1 on ESP32-C3 board and Expressif 3.0.0.0-a runtime with Core Info level diagnostics

shlomozippel, 2024-02-18:
Unfortunately it seems like the C2 doesn't have the RMT peripheral

dhalbert, 2024-02-18:
@rcmontrose Did you mean ESP32-C2 (not C3) in your post?

rcmontrose, 2024-02-18:
Sorry Dan, I did mention ESP32-C3 in the post but mislabeled the title. I've corrected the title to ESP32-C3.