adafruit/Adafruit_NeoPixel #373: ESP32-C3 "invalid interrupt priority"

aaroneiche, 2023-11-15:
- Arduino board:  ESP32-C3 based Boards (tested on an ESP32-C3 Devkit, but I'm willing to bet it affects the QT Py ESP32-C3)

- Arduino IDE version:  2.2.1
- ESP32 Library version 3.0.0-Alpha

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

* Open Examples -> Adafruit Neopixel -> strandtest
* Change pin/number of LEDS if neccesary
* Upload sketch
Sketch running produces error similar to:
`rmt: rmt_new_tx_channel(210): invalid interrupt priority`

And either repeats in a loop, or resets and the same error occurs.

UnexpectedMaker, 2023-11-20:
Yeah, RMT has been refactored in the 3.0.0 core and it's not backward compatible :(

JLeising, 2023-12-02:
Using Arduino library manager to revert to Adafruit Neopixel version 1.1.0 corrected this issue for me.

UnexpectedMaker, 2023-12-02:
Good to know!
There's an upstream fix from Espressif coming for the ESP32 3.0.0 core that should rectify this without Adafruit needing to deal with it. Not sure when that's landing though.

chibiconsulting, 2024-03-12:
FYI,  v1.11.0 seems to work for me. Was 1.1.0 a typo?

JLeising, 2024-03-13:
Not a typo. When I tried, most recent Adafruit Neopixel version did not work. Old version did. Maybe fixed now? Haven’t retested.

chibiconsulting, 2024-03-13:
No problem, I noticed it with 1.12.0 which I think is current; then tried v1.1.0 and it had other issues so I started again with 1.11.0 and it seems to work for me. I will keep watching as it would be nice to know when the current versions from Espresssif and Adafruit are on the same page again. Thanks.

egnor, 2024-08-05:
I _think_ the situation was something like
- arduino-esp32 v3 came out, using ESP-IDF v5.1
- Adafruit NeoPixel's use of the RMT libraries didn't work with ESP-IDF v5.1
- Adafruit patched it to work, but the fix runs out of stack with >70ish pixels (https://github.com/adafruit/Adafruit_NeoPixel/issues/375)
- MEANWHILE Espressif released newer libraries that fixed the original RMT issue
- so currently the latest Adafruit NeoPixel library does NOT work with arduino-esp32 v3, but earlier Adafruit library versions DO work on arduino-esp32 v3???

There's some guesswork in here...