adafruit/Adafruit_NeoPixel #375: pixel.show() crash with more than 73 pixel on ESP32s3

tresler, 2023-12-17:
I have weird problem. I test [example code](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino) or [example with new operator](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple_new_operator/simple_new_operator.ino) and everything works fine if I have 73 or less RGB leds. If I use 74 or more, everything crash. Can anyone explain it?

Debug code: [adafruit_neopixel_crash.txt](https://github.com/adafruit/Adafruit_NeoPixel/files/13697955/adafruit_neopixel_crash.txt)

- Arduino board:  ESP32s3-WROOM
- Arduino-ESP32: 3.0.0 alpha 3
- Arduino IDE: 2.2.1
- Adafruit_Neopoxel: 1.12

ajkeeton, 2024-06-05:
Same problem here, also Esp32 (Heltec Wifi Kit), and at around 70 pixels

teknynja, 2024-06-07:
I'm experiencing the same issue as well (crash/boot loop when driving more than around 70 pixels). This occurs on both my XIAO ESP32C3 boards and my ESP-WROOM-32 Dev Modules, with the following hardware/software:

* Boards: XIAO ESP32C3, Generic ESP-WROOM-32 38 Pin Dev Board
* Board Package: esp32 by Espressif (both v3.0.0 & v3.0.1)
* Adafruit NeoPixel v1.12.2
* Arduino IDE: 2.3.2

Digging around in the Adafruit NeoPixel Library, using the v3.x board package means were also using IDF v5, which is pretty simple, and the NeoPixel Library doesn't appear to be to blame for this issue. I've isolated the crash to where it's calling: `rmtWrite(pin, led_data, numBytes * 8, RMT_WAIT_FOR_EVER);`. The crash happens somewhere in there. Digging a little deeper, it doesn't look like the problem is in the IDF library either, since it doesn't really do much with the requested buffer size except to pass it along to the `rmt_transmit` method of the SDK. I expect somewhere downstream there a buffer is overrun and it eventually overwrites something important causing the whole thing to crash and reboot. 70 pixels turns out to be a fair amount of data: 70 pixels * 3 bytes for RGB, then each byte requires 8 RMT Items which themselves take 4 bytes each - 6720 bytes! This is far more than the ESP32-S3's 2048 bytes of RMT block memory (or the ESP32-C3's meager 768 bytes), so the SDK must be handling the buffering itself.

I'm not sure at this point where to turn for a fix though, since it looks like this is a problem with Espressif's v3 SDK. I'd really like to get this fixed, as I have a few projects with 150 or more pixels that I currently can't complete with this setup. I'm sure there's plenty of other people as well that expect to be able to drive larger pixel strings, so it'd be nice to get this working for everyone on the ESP32 platform.

teknynja, 2024-06-09:
So I posted about this issue on the Espressif forums (https://www.esp32.com/viewtopic.php?f=13&t=40270) and they quite helpfully pointed me in the right direction; it appears that the culprit in the NeoPixel library is the `espShow()` method in the `esp.c` file is allocating the RMT Items Buffer (`led_data`) on the stack, and at about 70ish pixels it exceeds the available stack space and causes the crash. I'm guessing that allocating that buffer from the heap (or statically) would likely greatly increase the number of pixels that can be driven.

While looking at this code more closely, I also noticed that it's calling `rmtInit()` every time the pixels are "shown" - I'm assuming that it probably should only be called once (maybe in the `.begin()` method?). There doesn't appear to be a close/cleanup function in the Adafruit_NeoPixel driver, so there's probably no place or need to call the matching `rmtDeinit()` to release the RMT channel.

teknynja, 2024-06-09:
@robertlipe - I'm definitely not an expert C/C++ coder, but to my eye

```c
void espShow(uint8_t pin, uint8_t *pixels, uint32_t numBytes, boolean is800KHz) {
  rmt_data_t led_data[numBytes * 8]; // <== stack allocation??

  if (!rmtInit(pin, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
    log_e("Failed to init RMT TX mode on pin %d", pin);
    return;
  }

  ...
```

the rather large `led_data` array is being allocated on the stack (plus in my demo code, once I change that to a static allocation outside the function, I'm able to send data to several hundred pixels without it crashing).

I haven't had a chance yet to try rigging up a test with allocating that buffer from heap in the actual driver, but I expect that should at least clear up the crashing issue.

teknynja, 2024-06-09:
I've submitted a pull-request with fixes for this issue. I'm able initialize the driver configured for 500 pixels and it seems to be working fine. Anyone who would like to try this code out can grab the changes at [teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fix](https://github.com/teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fixl) - let me know if it works for you or if you encounter any problems. Thanks again for the help from @robertlipe on getting me started on this patch.

Pr77Pr77, 2024-06-12:
@teknynja Your code works in my case. Thank you and @robertlipe very much. Also the link seems to be dead. This link to teknynja's fixes should work: https://github.com/teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fix

teknynja, 2024-06-12:
For anyone using this library in multi-threaded scenarios, I've created a new branch with the same fix but it also includes a mutex to make it thread safe (works just fine for single-threaded apps as well): [teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fix_safe](https://github.com/teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fix_safe). This is the preferred update for both single and multi-threaded applications.

teknynja, 2024-06-12:
Yeah, I know - I couldn't figure out a way to avoid it, but I figured that even if that happens, it will sort itself out - the first threads will just carry on and release their own locks (albeit unprotected), and then the last thread will set the mutex that everyone else uses from that point on.

teknynja, 2024-06-12:
Yeah, i wasn't sure how to avoid that without having a way to initialize with the object...

teknynja, 2024-06-13:
hmm, it doesn't appear we have access to the std library in this project (perhaps because this is just a plain ol `.c` file?). At least when I try to build in the Arduino environment, it complains that it can't find the header files. Also not sure how the maintainers would feel about adding references to std:: since it doesn't appear they are using it anywhere else.

Your example did get me looking into possible solutions from within the FreeRTOS framework itself, and I discovered the [`xSemaphoreCreateMutexStatic`](https://www.freertos.org/xSemaphoreCreateMutexStatic.html) function which looked _very_ promising, but digging through the FreeRTOS source code, it looks like that call is probably not be thread-safe either. :-/

I'm wondering if I could just do something quick-n-dirty, like disabling interrupts around where I'm checking/creating the mutex... it should be pretty quick through that section, and is in keeping with the design philosophy of this library.

I nags at me too, having even a tiny crack like this race condition in the code - my day job is working on industrial process control systems and we'd never let something like that into production.

Interestingly, I just started looking into c++ lambdas a short time ago to try and solve a different problem (I'm a big ~abuser~ user of lambdas in C# though!)

teknynja, 2024-06-13:
I was trying to avoid modifying the main `.cpp` file, but at this point it looks unavoidable. I could put the mutex in the initializer list, but that feels pretty invasive (not to mention how ugly the `#ifdef`s around the ctor would be) and it still doesn't completely close the race condition (just moves it to the ctors racing). Even if we don't do a "test and set", there's still an opportunity for a thread to stomp on another thread's mutex.

In a perfect world, the mutex would be created in the startup thread before any child threads are spawned, but that's not a reasonable thing to expect a typical Arduino user to deal with. The best we could probably hope for is to require the Adafruit_NeoPixel instances to be created on the main thread, and then calls to `.begin()`, `.show()`, et al could be done on the worker threads. Without low (cpu) level synchronization primitives, it think it's gonna be tough to create a mutex on a thread that needs to be protected with that very same mutex.

teknynja, 2024-06-15:
Updated pull request #394 to move the initialization of the `espShow()` mutext to the Adafruit_NeoPixel constructor. As long as all instances are created on the main (or same) thread, the race condition is avoided. Otherwise, there is a slight chance of the threads clobbering each other's mutex (the same behavior we had before this change).

This is a slightly more "invasive" change then before, as there are minor changes now to the `Adafruit_NeoPixel.cpp` & `.h` files :shrug:.

egnor, 2024-08-05:
I am just a kibitzer myself (who ran into this issue), but here's my understanding of the situation, for reference.

- The ESP32 Arduino core ["arduino-esp32"](https://github.com/espressif/arduino-esp32/) is built on top of ["ESP-IDF"](https://github.com/espressif/esp-idf/), an ESP32-specific support framework made by Espressif (makers of the ESP32). Any particular version of arduino-esp32 includes some version of ESP-IDF, and translates Arduino-type calls to ESP-IDF calls. (For example, when you call Arduino `digitalWrite`, arduino-esp32 calls ESP-IDF `gpio_set_level`.)
- Arduino code (sketches and libraries) can, and do, use the available version of ESP-IDF directly, since arduino-esp32 doesn't translate everything ESP-IDF does into Arduino-ish. You can "mix and match" Arduino-ish calls and ESP-IDF calls. Of course, if you use ESP-IDF, your code is necessarily ESP32 specific.
- On the ESP32, the Adafruit NeoPixel library, like other LED-strip-driving libraries, uses [the ESP32's "RMT" hardware module](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#section.15). This module is designed for driving infrared remote controls (like, controlling a TV) but is kind of a general timed pulse train synthesis module and can be pressed into service to drive NeoPixel-type LED strips with hardware, with guaranteed timing and without tying up the CPU "bit banging" a GPIO. Cool beans.
- The [arduino-esp32 RMT driver](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/rmt.html) doesn't offer quite enough control to efficiently drive LED strips (more on this later), so the Adafruit NeoPixel library uses the [ESP-IDF RMT driver](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html) directly. RMT is ESP32-specific anyway so this is fine. (On other chips, the NeoPixel library uses entirely different techniques.)

So far this is all fine. But we have some wrinkles:

- The previous [arduino-esp32 v2.0.14](https://github.com/espressif/arduino-esp32/releases/tag/2.0.17) was based on [ESP-iDF v4.4.7](https://github.com/espressif/esp-idf/releases/tag/v4.4.7), but the new [arduino-esp32 v3.0](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0) uses [ESP-IDF v5.1.4](https://github.com/espressif/esp-idf/releases/tag/v5.1.4).
- The ESP-IDF RMT driver interface [was completely rewritten between v4.x to v5.x](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/peripherals.html#rmt-driver). That potentially completely breaks the Adafruit code that uses the old ESP-IDF RMT driver interface.
- Good news: The "legacy" RMT driver is still available for compatibility, as of ESP-IDF 5.1! (It gets removed in later ESP-IDF versions, but arduino-esp32 isn't there yet.)
- Bad news: ESP-IDF will [deliberately crash on startup](https://github.com/espressif/esp-idf/blob/5ca9f2a49aaabbfaf726da1cc3597c0edb3c4d37/components/driver/deprecated/rmt_legacy.c#L1405) if both the legacy RMT library and newer RMT library are included in the firmware (even if both are not necessarily used).
- Worse news: In many cases, arduino-esp32 itself uses the newer RMT library (see https://github.com/espressif/arduino-esp32/issues/9866 for a deep dive).

To deal with this RMT driver interface change in ESP-IDF v5.x, Adafruit (Lady Ada herself!) [changed the NeoPixel library](https://github.com/adafruit/Adafruit_NeoPixel/pull/367). On ESP-IDF v5.x, instead of using the ESP-IDF RMT driver, it uses the arduino-esp32 wrapper driver, which in arduino-esp32 v3.x uses the new ESP-IDF RMT driver.

BUT, there was a reason to bypass arduino-esp32 originally. (Remember I said I'd get to this.) The ESP32 RMT hardware can be programmed in various ways. One is just to give it a whole sequence of exact bit timing in a big buffer. (Like, "set the line HIGH for 350 microseconds, then LOW for 75 microseconds, then...".) This is the only mode the arduino-esp32 wrapper supports. But there's another mode which ESP-IDF supports, where you store the buffer in your own favorite format, and supply a little translation function that turns a chunk of buffer into a chunk of timed bits, and ESP-IDF calls that function from an interrupt as needed to translate a chunk at a time, which is much more memory efficient. This "streaming" mode is what the NeoPixel library uses on ESP-IDF v4.x. On v5.x, with the Arduino wrapper, it just allocates a (potentially) big buffer -- 32 bytes of buffer for every byte of pixel data! And it does so on the stack.

(Note the ironic-in-hindsight exchange in https://github.com/adafruit/Adafruit_NeoPixel/pull/367 where @ladyada wonders if it's OK to allocate the buffer on the stack, https://github.com/me-no-dev notes that it could be pretty big and should maybe be allocated on the heap, but @ladyada said "the stack memory is fine", probably thinking about post-return references and not about stack exhaustion...)

- But, good news! The arduino-esp32 maintainers [added a way to not use the newer RMT library](https://github.com/espressif/arduino-esp32/pull/9941), allowing the legacy RMT library to be used. This got included in arduino-esp32 v3.0.3.
- Mediocre news! The way you do that disabling is kinda hinky, you have to create a "header file" with the magical name `build_opt.h`, which isn't a header file at all but a list of compiler flags, and add `-DESP32_ARDUINO_NO_RGB_BUILTIN` to that.
[... cut ...]
Going forward, probably the NeoPixel library should have code that uses the new ESP-IDF RMT driver, which does still support streaming mode (but with a whole different interface). Then everything will be dunky hory, though that will be a bunch of work for someone to do.

Meanwhile, allocating that big buffer on the heap is probably a good move. It's not a cure-all, though -- depending on the length of your pixel chain, you could end up running out of heap memory! (1000 pixels * 3 bytes/pixel * 32x blowup = 96KB, which is a lot of memory for an ESP32.)

teknynja, 2024-08-05:
@egnor – I think that captures (in great detail) the situation surrounding this issue, thanks! And @robertlipe, I had no idea about the drama/issues going on with Platformio (ignorance is bliss!).

I agree that the amount of buffer space per NeoPixel is problematic, the primary motivation for my work on this issue was to be able to break the unexpectedly low limit of ~73 pixels when I was using the new ESP-IDF RMT driver for other purposes in my project as well. Initially, I attempted to use the low-level solution of translating the pixel buffer data "on-the-fly", but getting the timing right to where that translation code was called with consistently low enough latency to keep the RMT’s buffers fed proved challenging. If I remember, it also caused problems with the 5.x RMT library’s buffer allocation, which otherwise seems to work very well as long as you stick to the 5.x API.

I think until a better solution comes along, the 1000ish pixel limit will get most users of this library where they want to go, and if it doesn’t then they will need to select a different platform to achieve their goals (just like you would need to step up from the ATmega328 if you wanted to drive more than a few hundred pixels). I might be a bit biased about getting my changes merged into this library, but I think given the state of things at the moment, this change allows a large percentage of users to get on with their development and worry about other problems.

Thank you for prodding this issue, I was beginning to wonder if it was destined to end up as a footnote on archive.org 😉

egnor, 2024-08-05:
@ladyada take a look at the two PRs (pick one or the other, not both) by @teknynja
- https://github.com/adafruit/Adafruit_NeoPixel/pull/392
- https://github.com/adafruit/Adafruit_NeoPixel/pull/394

(My guess: Having now researched the background, probably you'd like to keep the behavior of initializing and deinitializing RMT entirely within `show()` rather than doing what that PR does, but that's a quick change if desired. In general the quickest of fixes would just be to `malloc()` the buffer on entry to the function and `free()` it on exit, that would be the most surgical fix though not the most performant.)

teknynja, 2024-08-05:
@ladyada - [RMT Buffer Allocation Fix (Thread-Safe) for Issue #375 #394](https://github.com/adafruit/Adafruit_NeoPixel/pull/394)  is the PR I recommend as includes tweaks to improve the thread-safety of the fix. The other PR tries to keep all the changes within the `show()` method at the expense of some thread safety.

And yes @ladyada, get some rest!

teknynja, 2024-08-05:
The "bitbucket" test is a valid way to make sure the code compiles and runs w/o crashing, but in order to make sure there aren't any visual glitches or other timing artifacts, we'd at least need to connect a reasonable number of actual devices and watch the result (or pour over a logic analyzer's output ;-) )

zackees, 2024-08-06:
QEMU based esp32 simulator:

https://github.com/mluis/qemu-esp32

@robertlipe

egnor, 2024-08-06:
I actually did the exercise of writing a new-RMT-interface-based streaming LED strip driver, cribbing from [Espressif's led_strip component](https://components.espressif.com/components/espressif/led_strip). There's some subtleties (like state management in the encoder function) but it's not too complicated. So @robertlipe wrapping Espressif's component as the "backend" seems reasonable, or maybe adapt that code to something a little more backend-able (e.g. no need to keep its own pixel buffer).

As for automated testing, I suspect you're not _entirely_ serious about that particular goose chase, but I would have done it with automation on top of a logic analyzer (or maybe scope). The kind you'd need would be expensive, but...

zackees, 2024-08-21:
FastLED dev here.

I'm looking at the ESP32 documentation on the new RMT changes.

Some changes aren't that big of a deal. Others are painful.

It's **possible** that a subset of features that intersect between the esp-idf RMT versions can be merged together into a unified driver that somewhat resembles the legacy driver. What makes this easier is that the led devs only need to be concerned about the TX portion of the RMT protocol.

The big changes I see are the following:
  * ISR callbacks
  * Calling status and other functions after the channel is created is now a no-no. It can only be set during construction time of the RMT driver (set clock divisor / gpio).
  * Streaming support for RMT symbol writing has been changed from explicit looping on an ISR to implicit callbacks managed by the driver.

I haven't dug too much further than the docs, but it could be possible to create a unified driver where led devs only need to change it minimslly once for the 4.x codebase and then it just works in the 5.x codebase. Things like dynamic status and pin changing would be disabled in this unified driver but I don't think that's a big problem and could be bridged with a strategy where if you want to change pins you just re-initialize the whole channel.

Just my 2c.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/peripherals.html#rmt-driver

[image]
[image]

tyeth, 2025-01-14:
Please retest with latest release (v1.12.4)

MrBroccoliJP, 2025-02-02:
I suspect my issue is related to this so I'm gonna leave my feedback here. Please feel free to request any additional information. Aspiring junior dev here so I'm sorry in advance for the lack of any info,

I have a neopixel clock which the code works fine for Esp32 core <v3.0.0.
I tested the latest neopixel release,  it no longer crashes immediately after uploading the sketch.

This specific function works fine in Esp32 Core 2.0.17:
[fades the dots of the clock in and out every second, rough code not optimized wip]

```
void fade_tick(short int x_loc, short int y_loc_led1, short int y_loc_led2) {

  short int correction;

  if ((millis() - tmp_millis) > 500 && (millis() - tmp_millis) <= 1000) {
    if (r_tmp != r) {
      r_dec = double(r) / 23;
      r_tmp = (r_tmp + r_dec <= r) ? (r_tmp + r_dec) : r;
    }
    if (g_tmp != g) {
      g_dec = double(g) / 25;
      g_tmp = (g_tmp + g_dec <= g) ? (g_tmp + g_dec) : g;
    }
    if (b_tmp != b) {
      b_dec = double(b) / 24;
      b_tmp = (b_tmp + b_dec <= b) ? (b_tmp + b_dec) : b;
    }
    //fade in tick until 500miliseconds (1/2 second)
  } else if ((millis() - tmp_millis) < 500 && (millis() - tmp_millis) <= 1000) {
    if (r_tmp != 0) {
      r_dec = double(r) / 25;
      r_tmp = (r_tmp - r_dec >= 0) ? (r_tmp - r_dec) : 0;
    }
    if (g_tmp != 0) {
      g_dec = double(g) / 23;
      g_tmp = (g_tmp - g_dec >= 0) ? (g_tmp - g_dec) : 0;
    }
    if (b_tmp != 0) {
      b_dec = double(b) / 24;
      b_tmp = (b_tmp - b_dec >= 0) ? (b_tmp - b_dec) : 0;
    }
    //fade out tick until 1000miliseconds (full second)
  } else if (((millis() - tmp_millis) > 1000) && ((millis() - tmp_millis) < 1999)) {
    //in this case tmp_millis might be misaligned with the seconds, so this tries to correct it.
    correction = 1000 - (millis() - tmp_millis);
    tmp_millis = millis() - correction;
  } else if ((millis() - tmp_millis) >= 2000) {
    tmp_millis = millis() + country.ms();  //correct it with ms from eztime
  }

  uint16_t color = matrix.Color(int(r_tmp), int(g_tmp), int(b_tmp));
  matrix.drawPixel(x_loc, y_loc_led1, color);
  matrix.drawPixel(x_loc, y_loc_led2, color);
  matrix.show();
}
```

With the previous neopixel version and Esp32 Core >v3  it crashed immediately after uploading, 

Now with this latest neopixel update v1.12.4 and Esp32 Core v3.1.1 the code runs but the fade is no longer visible, or the dots get stuck in a certain state. Basically it seems that matrix.show() is not pushing the information like it should.

Reverted back to the ESP32 Core v2.0.17, and it works fine.

dhalbert, 2025-02-02:
@MrBroccoliJP I have posted your issue in #420. In general it's better to open a new issue when an issue has been closed, but there seems to be a related problem, unless the original problem itself is not fixed.

mikeysklar, 2026-09-17:
Retested on a Metro ESP32-S3 with arduino-esp32 3.3.10 (IDF 5.5.4). Nothing was attached to the data pin, since the crash only depends on the pixel count. The sketch calls `show()` 200 times at each count from 1 to 1000, and I ran each library version on 10 separate power-ups.

| Library | Result, 10 power-ups | Last OK count | Stack headroom at 72 pixels |
|---|---|---|---|
| 1.12.3 | 10/10 panic reboot at 73 pixels | 72 | 136 B |
| 1.15.5 | 10/10 pass | 1000 | 6,332 B |

#394 (first released in 1.12.4) moved the RMT buffer from the stack to the heap, and that fixed this. With the fix, stack use stays flat from 1 to 1000 pixels. I think this can be closed. #420 is a different symptom and stays open.

closed 2026-09-17 by mikeysklar