adafruit/Adafruit_NeoPixel #378: trip.show will cause esp8266 wdt reset.

chuz, 2024-01-25:
device stuck. help

log:
19:42:42.968 -> ---2195178
19:42:50.027 -> 
19:42:50.027 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,5)
19:42:50.027 -> 
19:42:50.027 -> wdt reset
19:42:50.027 -> load 0x4010f000, len 3424, room 16 
19:42:50.027 -> tail 0
19:42:50.027 -> chksum 0x2e
19:42:50.027 -> load 0x3fff20b8, len 40, room 8 
19:42:50.027 -> tail 0
19:42:50.027 -> chksum 0x2b
19:42:50.027 -> csum 0x2b
19:42:50.027 -> v00044340
19:42:50.027 -> ~ld
19:42:50.192 -> 
19:42:50.673 -> .......---4400
19:42:55.409 -> ---5401

`#include <Adafruit_NeoPixel.h>
#include <WebSocketsServer.h>
#ifdef __AVR__
  #include <avr/power.h>
#endif

#define PIN 3

const char* ssid = "fornut";
const char* password = "a11111111";
IPAddress Ip(192,168,2,10); // IP address for your ESP
IPAddress Gateway(192,168,2,1); // IP address of the access point
IPAddress Subnet(255,255,255,0); // subnet mask
WebSocketsServer webSocket = WebSocketsServer(81);

Adafruit_NeoPixel strip = Adafruit_NeoPixel(180, PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  // This is for Trinket 5V 16MHz, you can remove these three lines if you are not using a Trinket
  #if defined (__AVR_ATtiny85__)
    if (F_CPU == 16000000) clock_prescale_set(clock_div_1);
  #endif
  // End of trinket special code
  Serial.begin(115200);
  WiFi.setOutputPower(15);
  WiFi.config(Ip, Gateway, Subnet);
  WiFi.begin(ssid, password);
  Serial.println("");
  while (WiFi.status() != WL_CONNECTED) { 
    delay(500);
    Serial.print(".");
  }
  strip.begin();
  strip.setBrightness(50);
  strip.show(); // Initialize all pixels to 'off'
}

unsigned long currentTime = millis();
unsigned long currentTimePrint = millis();

uint16_t j = 0;

void loop() {
  if (millis() - currentTime > 3) {
    uint16_t i;
      for(i=0; i< strip.numPixels(); i++) {
        strip.setPixelColor(i, Wheel(((i * 256 / strip.numPixels()) + j) & 255));
      }
      strip.show();
      
      currentTime = millis();
      if (j++ > 255*5) {
        j = 0;
      }
  }

  if (millis() - currentTimePrint > 1000) {
    Serial.printf("---%d\n", millis());
    currentTimePrint = millis();
  }
}

uint32_t Wheel(byte WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return strip.Color(255 - WheelPos * 3, 0, WheelPos * 3);
  }
  if(WheelPos < 170) {
    WheelPos -= 85;
    return strip.Color(0, WheelPos * 3, 255 - WheelPos * 3);
  }
  WheelPos -= 170;
  return strip.Color(WheelPos * 3, 255 - WheelPos * 3, 0);
}`

caternuson, 2024-01-25:
Try commenting out this line:
```cpp
    delay(500);
```

chuz, 2024-01-26:
> Try commenting out this line:
> 
> ```c++
>     delay(500);
> ```

Which line...? In setup?

chuz, 2024-01-26:
> Try commenting out this line:
> 
> ```c++
>     delay(500);
> ```

In setup? When I test, the code is running about 30min and then got stuck.  The device does not get stuck as soon as it is turned on

caternuson, 2024-01-26:
Do you see the same behavior running the example from the library?
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino

In general, need to come up with an example sketch that is NeoPixel only (no WiFi or WebSocketsServer) that demonstrates this issue.

Simon-lincz, 2024-01-29:
useing FASTLED...

closed 2024-01-29 by Simon-lincz