adafruit/Adafruit_NeoPixel #379: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'  IRAM_ATTR void espShow(

yunshang6, 2024-01-28:
编译时报错:
`
D:\ProgramData\xx\Arduino\libraries\Adafruit_NeoPixel\esp8266.c:20:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 IRAM_ATTR void espShow(
           ^
exit status 1
为开发板 NodeMCU 0.9 (ESP-12 Module) 编译时出错。
`

uushaggy, 2024-08-25:
same here:
/usr/local/bin/arduino-cli compile --fqbn esp8266:esp8266:nodemcuv2:baud=115200,xtal=80,eesz=4M,dbg=Disabled,lvl=None____,ip=lm2f,vt=flash,exception=disabled,wipe=none --build-cache-path /tmp --output-dir /tmp/2267355810/build --build-path /tmp/arduino-build-B9595D3CF2719D8867E5E8A64F91B25C  /tmp/2267355810/simple

/home/builder/Arduino/libraries/adafruit_neopixel_1_12_3/esp8266.c:20:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 IRAM_ATTR void espShow(
           ^
/home/builder/Arduino/libraries/adafruit_neopixel_1_12_3/Adafruit_NeoPixel.cpp:387:12: error: 'IRAM_ATTR' does not name a type
 extern "C" IRAM_ATTR void espShow(uint16_t pin, uint8_t *pixels,
            ^
/home/builder/Arduino/libraries/adafruit_neopixel_1_12_3/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
/home/builder/Arduino/libraries/adafruit_neopixel_1_12_3/Adafruit_NeoPixel.cpp:3163:42: error: 'espShow' was not declared in this scope
   espShow(pin, pixels, numBytes, is800KHz);
                                          ^
Multiple libraries were found for "Adafruit_NeoPixel.h"
  Used: /home/builder/opt/libraries/adafruit_neopixel_1_12_3
  Not used: /home/builder/opt/libraries/rakwireless_led_matrix_1_0_0
  Not used: /home/builder/opt/libraries/pinduino_1_5_0
Error during build: exit status 1

uushaggy, 2024-08-25:
Cannot compile a script using ADAFruit_Neopixel.h library on Cloud editor. Works on local editor.

zweisteinko, 2024-12-05:
this problem on all versions arduino
1.8.18
2.3.4
same error