adafruit/Adafruit_NeoPixel #380: Watchdog timeout reset when using the expressif esp32 board package 3.0.0-alpha

jens-j, 2024-02-23:
- Arduino board:  ESP32 Dev Module (ESP32-WROOM)
- Arduino IDE version: 2.3.2

After updating my esp32 board package from 2.0.11 I noticed that using the `Adafruit_NeoPixel::show` function results in a TG1WDT_SYS_RESET above a certain number of pixels. I can reproduce the issue using the `simple` example from the Adafruit Neopixel library. The sketch works correctly with the default number of 16 pixels. But when I increase it to 100, the watchdog timeout occurs during the call to `show()`. This lead to a boot loop where the device continuously resets and prints the message shown below to the serial interface.

So to be clear, this is issue only occurs when doing this in combination with the alpha build of the board package. The stable 2.0.11 release works fine. The reason I am looking at the 3.0.0 package is because it includes a more powerful set of ADC related functions beyond the standard `analogRead`.

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c`

Pr77Pr77, 2024-06-09:
I have the same problem. This is my output stating at the call of show():

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x40084748  PS      : 0x00060936  A0      : 0x80084820  A1      : 0x3ffb0330  
A2      : 0x00000054  A3      : 0x00000804  A4      : 0x00000000  A5      : 0x3ffb02ec  
A6      : 0x3ffc4624  A7      : 0x00000030  A8      : 0x00000000  A9      : 0x3ffb1c70  
A10     : 0x3ffc39fc  A11     : 0x0000012b  A12     : 0x00000000  A13     : 0x40393000 
A14     : 0x00000002  A15     : 0x00000009  SAR     : 0x0000001a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400860e9  LEND    : 0x400860f1  LCOUNT  : 0x00000027  

Backtrace: 0x40084745:0x3ffb0330 0x4008481d:0x3ffb0360 0x40093c78:0x3ffb0380 0x40090ad5:0x3ffb03a0 0x40090d28:0x3ffb03c0 0x400e8f14:0x3ffb03e0 0x400e31f5:0x3ffb0430 0x400e30fb:0x3ffb1c70 0x400d392a:0x3ffb1c90 0x400d7792:0x3ffb1ce0 0x400d7c22:0x3ffb1d70 0x400d8a15:0x3ffb1fb0 0x400da509:0x3ffb21f0 0x400eaeef:0x3ffb2290

ELF file SHA256: 2de4ef93290321d2

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1448
load:0x40078000,len:14844
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
```
Can please someone help?

egnor, 2024-08-05:
This is probably the same as https://github.com/adafruit/Adafruit_NeoPixel/issues/375?