adafruit/Adafruit_NeoPixel #383: not working with AMB82-MINI

rkuo2000, 2024-03-19:
I have bought Neopixel ring and matrix, both are not working with AMB82-MINI (RTL8735B),
https://www.amebaiot.com/en/amebapro2/
it would be good to see Neopixel working with this EdgeAI MCU (WiFi, BLE4.2, 0.4TOPs NN engine)
work as portable ChatGPT device.

caternuson, 2024-03-19:
What does "not working" mean? Please provide more details.

rkuo2000, 2024-03-19:
> What does "not working" mean? Please provide more details.

No dots（led）will light up, I tried 1,4,8,16 LEDs , but 1st led will be on if use use resistor to touch it for pullup. I saw esp8266.cpp & esp.cpp are in the Adafruit Neopixel library, but I can't get it running either, saw some post about Neopixel need <1.5us pulse when use GPIO to drive Neopixel

rkuo2000, 2024-03-19:
Adafruit Neopixel Library/examples/strandtest/strandrest.ino

caternuson, 2024-03-19:
The pinout and datasheet information for that board does not appear to be publicly available. Do you have its pinout information?

rkuo2000, 2024-03-19:
pin is Vcc (5V), Din, Gnd at the Neopixel device, 
sample code assign a GPIO  to send 0/1 to Din
please refer to the code in [https://github.com/adafruit/Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel)

caternuson, 2024-03-19:
Which pin on the AMB82-MINI is the NeoPixel's DIN connected to?

Are you changing the code in the strandtest.ino to match the pin being used?

Without the actual pinout information for the AMB82-MINI, how are you determining what pin should be used?

rkuo2000, 2024-03-20:
AMB82-mini pinout is [here](https://www.amebaiot.com/wp-content/uploads/2022/12/amb82-mini/P02.png)
I just assign one GPIO pin to drive WS2812B LED's DIN

caternuson, 2024-03-20:
>  I just assign one GPIO pin to drive WS2812B LED's DIN

Which pin?

rkuo2000, 2024-03-20:
https://github.com/rkuo2000/Arduino/blob/master/examples/AMB82-MINI/LED5050_WS2812B/LED5050_WS2812B.ino

rkuo2000, 2024-03-20:
I think this Adafruit library wont work, not even using ESP32, ESP8266, I am trying other WS2812B libraries.

caternuson, 2024-03-20:
This library is known to work fine on ESP32, ESP8266, and others.

If it is not working for your specific setup, then more information is needed. At this point, it is still not clear if the correct pin is even being used and referenced correctly in code.

rkuo2000, 2024-03-20:
https://www.amebaiot.com/en/amebad-arduino-ws2812b-basics/
I will try this one first. Thanks !

rkuo2000, 2024-03-20:
> This library is known to work fine on ESP32, ESP8266, and others.
> 
> If it is not working for your specific setup, then more information is needed. At this point, it is still not clear if the correct pin is even being used and referenced correctly in code.
> 
> The code linked has this:
> 
> ```c++
> #define PIN        1 // On Trinket or Gemma, suggest changing this to 1
> ```
> 
> Is this the specific pin the NeoPixel DIN is being connected? ![pin](https://private-user-images.githubusercontent.com/8755041/314561487-31c676ff-0b2d-437c-aaa6-ee496294ba94.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDg4NTcsIm5iZiI6MTcxMDk0ODU1NywicGF0aCI6Ii84NzU1MDQxLzMxNDU2MTQ4Ny0zMWM2NzZmZi0wYjJkLTQzN2MtYWFhNi1lZTQ5NjI5NGJhOTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjBUMTUyOTE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjRjNDM2YWQ2YTliMjMxZTFhYTY5YWI4NjRiZGRiNmM0NGU1MGZlOGM4ZTMzZTVhYTE3ODZiODFhZTk1YWE0OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.csSN6xMjhSteEtlwnMpt_0B21cX0Ncn1vxcSdSZlyMw)

Yes

caternuson, 2024-03-20:
That nominally seems OK in terms of pin set in code. It could be other things like power and logic level.

Try that other library and report back here if it was successful or not. If that library works, it would eliminate a lot of possible hardware related causes.

rkuo2000, 2024-03-21:
For BW16. WS2812B_Basics.ino works, I can set LED one by one using led.setPixelColor,
but led.fill failed to drive multiple leds.
WS2812B_Patterns.ino failed to drive leds.

caternuson, 2024-03-21:
So the issue is not specific to this library. You are also seeing issues using the amebaiot examples and library. Please open this issue in their forums: 
https://forum.amebaiot.com/

nsmith1024, 2024-07-05:
Hello,

Are the pins reversed on the diagram, or just the labels?

https://www.amebaiot.com/en/amebapro2-amb82-mini-arduino-getting-started/

Why would anybody show the pins and names on the BACK of the board? Who can see it if the board is showing the BACK?

If its indeed reversed, the is pin 0 really on the right/top of the board?

Something is wrong because i cant seem to get the correct GPIOS to turn on/off.

Also im pretty sure i burnt the thing out because i dont really know where the ground is because everything is reverse, and how can pin 0 be at the top/right.

Maybe some things are flipped and some not?

Thanks for the clarification.

Maybe someone can update the pinouts with the board FACING UP like a normal person would do it?

rkuo2000, 2024-07-06:
The pin out diagram on the left (BACK) is the printed on AMB82-mini board, you can turn it to see it on PCB.