adafruit/Adafruit_NeoPixel #391: With Arduino Nano ESP32 no display when using "By Arduino pin"

ednieuw, 2024-05-24:
- Arduino board:  Arduino Nano ESP32

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.3.2

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Both with SK6812 and WS2812 LEDs this issue occurs

How to test:
With RGBWstrandtest.ino the pin number changed to D5 and LED count to 14 (in my case)

```
// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN     D5

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT  14
```

When Pin numbering: By Arduino pin (default) --> LEDs will NOT turn on
When Pin numbering: By GPIO number (legacy)  --> LEDs turns ON

I expect that when using the D and A pin numbering in my sketch the pin number scheme irrelevant

See: https://support.arduino.cc/hc/en-us/articles/10483225565980-Select-pin-numbering-for-Nano-ESP32-in-Arduino-IDE

-------------
You can select pin numbering in Tools > Pin Numbering option:

By Arduino pin (default): Standard for Nano boards by Arduino.
By GPIO number (legacy): Standard for most ESP32 boards.
Regardless of which pin numbering you use, you can use the pin labels to refer to the pins. The following lines of codes are equivalent, but only the one using D0 to refer to the pin works with both pin numbering options:

pinMode(2, OUTPUT) (By Arduino pin)
pinMode(5, OUTPUT) (By GPIO number)
pinMode(D0, OUTPUT) (By Arduino pin or By GPIO number)
For a complete overview, see the table below.
----------------

caternuson, 2024-05-24:
> When Pin numbering: By Arduino pin (default) --> LEDs will NOT turn on
> When Pin numbering: By GPIO number (legacy) --> LEDs turns ON

This is a known "feature" specific to the Arduino Nano ESP32:
https://github.com/espressif/arduino-esp32/issues/9150

TLDR = Use the _By GPIO number (legacy)_ option

ednieuw, 2024-05-25:
Thanks for the reference. 
I only had a pin problem with the Neopixel library and not with LEDs on Digital pins pin D9, D10 and LEDBUILTIN D13 and the RTClib library (pins A4 and A5) and LDR (pin A2) that use the analogue pins. 

I use the Arduino Nano ESP32 board and not the ESP32 -ESP32S3 Dev board V2.0.16. The ESP32S3 Dev board board does not recognize Arduino's  D and A numbering.

You would expect the Arduino Nano ESP32 board compiler would work flawlessly.  
If you look in the Arduino IDE 2.3.2 the pin D5 translates to GPIO 8 as it should. 
Therefore I think the problem is in the Neopixel library

[image]

I prefer to use the default setting of the compiler as much as possible to avoid problems. Especially in this case because all works fine except the SK6812 or WS2812 strip