adafruit/Adafruit_NeoPixel #396: Arduino Nano Matter

JakobGlieder, 2024-07-01:
The library does not seem to work with the Arduino Nano Matter

acsandst, 2024-07-04:
following
Also interested in this to work ☺️
thanks!

tiziano149, 2024-09-01:
Bump and follow

tiziano149, 2025-02-10:
Bump too 😁

caternuson, 2025-02-11:
Until this library gets updated, it looks like there is core specific support via ezWS2812 as of the 2.2.0 release:
https://github.com/SiliconLabs/arduino/issues/56
https://github.com/SiliconLabs/arduino/releases/tag/2.2.0

Examples are here:
https://github.com/SiliconLabs/arduino/tree/main/libraries/ezWS2812/examples

It looks like it relies on using the SPI peripheral so would be limited to the SPI pin `MOSI`:
https://docs.arduino.cc/tutorials/nano-matter/user-manual/#pinout

JakobGlieder, 2025-02-11:
Please note that it also only works with the SPI Pin

tiziano149, 2025-02-12:
> RGBW is exactly what I was looking for. Is the adoption something you are willing to share? Anders.

Sure! Here is the RGBW-Version of the library:

[https://github.com/tiziano149/ezSK6812](url)

 #https://github.com/tiziano149/ezSK6812

acsandst, 2025-03-02:
Big thanks!

tiziano149, 2025-04-14:
Any support on this? Arduino Nano Matter is officially out.

ladyada, 2025-05-03:
PR's welcome 
bumps are not - we don't own this board and have no plans to write the bitbang code required.