adafruit/Adafruit_NeoPixel #401: strip.show(); crashes Arduino Nano ESP32 with Espressif ESP32 3.0.4 board

ednieuw, 2024-09-04:
- Arduino board:  Espressif ESP32 3.0.4

- Arduino IDE version:  1.8.19 and 2.3.2

Choose the Arduino Nano ESP32 from the Espressif esp32 board.
Partition Scheme: With FAT
Pin Numbering: By GPIO number (legacy)

Open examples-> Adafruit->RGBW strandtest
Compile and upload
The Nano ESP32 will crash immediately.

In my own program, that uses this library, the crash follows when using the strip.show(); command. The program starts up and runs well until the strip.show(); command is used and the MCU reboots.

NOTE:
When the Arduino ESP32 board is used the strip.show(); command works well.

Dcmwrnd, 2024-09-13:
Me too, using Arduino IDE v1.8.19 &  ESP32 Dev Module with Board Manager v3.0.4. Once strip.show(); is called the ESP32 reboot with error messages below. The same code compiled with Board Manager v2.0.17 without any problems.

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x400e52b3  PS      : 0x00060036  A0      : 0x800e53fe  A1      : 0x3ffaf0f0  
A2      : 0x00000016  A3      : 0x3ffb9d0c  A4      : 0x00000186  A5      : 0x3ffaf0f0  
A6      : 0x3ffb0320  A7      : 0x3ffb21b0  A8      : 0x00088004  A9      : 0x00000091  
A10     : 0x00008004  A11     : 0x00000004  A12     : 0x00000008  A13     : 0x00000008  
A14     : 0x00000004  A15     : 0x00000004  SAR     : 0x00000003  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400e5280  LEND    : 0x400e52b7  LCOUNT  : 0x00000003  

Backtrace: 0x400e52b0:0x3ffaf0f0 0x400e53fb:0x3ffb21f0 0x400d52a2:0x3ffb2210 0x400d5321:0x3ffb2230 0x400d5706:0x3ffb2250 0x400e967c:0x3ffb2270 0x40090f3e:0x3ffb2290

ELF file SHA256: 44314907c87f7e0b

ednieuw, 2024-09-23:
I wrote a library for the SK6812 and WS2812 LEDs  
https://github.com/ednieuw/EdSoftLED

ednieuw, 2024-12-10:
Still an issue with Neopixel version 1.12.3 and esp32 core 3.0.7 and 256 LEDs in strip

dhalbert, 2025-01-12:
Closing as a duplicate in factor of #402

closed 2025-01-12 by dhalbert