adafruit/Adafruit_NeoPixel #402: Neopixel crash with >75 pixels using ESP32 Core 3.0.x

Nourbakhsh-Rad, 2024-09-13:
I am running into an issue where after upgrading the ESP32 package from 2.0.17 to 3.0.x, Neopixel sketches with an LED count of 76 or more get into a crash loop

I am using the Adafruit_NeoPixel '[simple.ino](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/simple/simple.ino)' sketch for testing

Board: ESP32 Dev Module (ESP32-devkitC-v4)
Hardware Configuration: ESP32 Dev Module
Version: v3.0.4
IDE Name: Arduino IDE 2.3.2
Operating System: Windows 10
Flash frequency: 80
PSRAM enabled: no
Upload speed: 921600

Note that, I am using the latest version of the Adafruit_NeoPixel library and the Arduino ESP32 core 3.0.4
----------------------------------------------------------------
when using ESP32 package 2.0.17 everything is ok
with ESP32 package 3.0.x and LED_COUNT <76 everything is also ok

Similar issue reported in https://github.com/adafruit/Adafruit_NeoPixel/issues/401#issue-2505462349 and https://github.com/espressif/arduino-esp32/issues/9903#issue-2362334081 and https://github.com/espressif/arduino-esp32/pull/9906#issue-2363288523 and [here](https://forum.arduino.cc/t/neopixel-crash-with-75-pixels-using-esp32-core-3-0-x/1273500/4)

imjustger, 2024-09-20:
I encountered the same issue. My project, which uses a XIAO ESP32S3 board running ESP32 2.0.17 by Espressif Systems, worked fine until I updated the firmware yesterday. Without noticing, I upgraded to Arduino’s 3.0.5 version for the board, which now causes my microcontroller to crash whenever the Neopixel library is invoked.”The microcontroller crashes and triggers a stack overflow error (***ERROR*** A stack overflow in task loopTask has been detected.) when running even the most basic NeoPixel control sketches.

Steps to Reproduce:
	1.	Set up an LED strip (SK6812 RGBW) with the XIAO ESP32S3.
	2.	Install NeoPixel library version 3.0.5.
	3.	Upload a basic sketch controlling the LED strip (e.g., a simple color wipe).

Expected Behavior:

The LEDs should work without issue, and the system should remain stable.

Actual Behavior:

	•	The ESP32S3 microcontroller experiences a stack overflow error, causing crashes.
	•	The error is consistently reproducible with version 3.0.5.

Temporary Fix:

Rolling back to NeoPixel v2.0.17 resolves the issue entirely. With version 2.0.17, the microcontroller operates as expected, without crashing or showing any signs of instability.

Environment:

	•	Board: XIAO ESP32S3
	•	NeoPixel Library: v3.0.5 (issue), v2.0.17 (working)
	•	Arduino IDE: 2.3.2
	•	LED Strip: SK6812 RGBW

It seems that version 3.0.5 of the NeoPixel library has compatibility or memory management issues with the XIAO ESP32S3, specifically leading to a stack overflow. Please investigate the changes in v3.0.5 that might be causing this issue. I’m happy to provide additional details if needed.

Nourbakhsh-Rad, 2024-09-20:
The problem seems to be with the **esp32-hal-rmt** library of the new ESP32 core (3.0.5)

sanchosk, 2024-10-01:
For SK6812 LED strips with white leds, the library fails the same way after 56 LEDs.

k11o, 2024-10-29:
I've got same issue. And I dealed this with [this forked library](https://github.com/teknynja/Adafruit_NeoPixel/tree/esp32_rmt_memory_allocation_fix_safe).

plin2, 2024-10-29:
I went a completely different way. For what I need WLED is doing the job.

Nourbakhsh-Rad, 2025-01-12:
### _The problem still persists!_

Board: ESP32 Dev Module (ESP32-devkitC-v4)
Hardware Configuration: ESP32 Dev Module
Version: v3.1.1
IDE Name: Arduino IDE 2.3.4
Operating System: Windows 10
Flash frequency: 80
PSRAM enabled: no
Upload speed: 921600

Nourbakhsh-Rad, 2025-01-18:
**_The problem is solved_**
Board: ESP32 Dev Module (ESP32-devkitC-v4)
Hardware Configuration: ESP32 Dev Module
Arduino ESP32 Core: v3.1.1
_Adafruit_NeoPixel: 1.12.4_
IDE Name: Arduino IDE 2.3.4
Operating System: Windows 10
Flash frequency: 80
PSRAM enabled: no
Upload speed: 921600

closed 2025-01-18 by Nourbakhsh-Rad