adafruit/Adafruit_NeoPixel #410: STM32 (RAK3172) and WS2816C-1313 problems

Jan-OleGiebel, 2024-11-06:
Hi!
I'm currently struggling with getting WS2816C-1313 neopixels to work with a STM32WL (RAK3172).
When specifying the correct numer of pixels (16) only 8 light up and only turn on with the color white, no other color works.
Sometimes the last LED turns green. When I set the number of LEDs to 32 all light up. However they flicker and I'm unable to set any color.
I've tried different speeds, etc. But nothing seems to fix the issue.
Do you have any suggestions, how I can get the LEDs working?

Thanks a lot!

Jan-OleGiebel, 2024-11-10:
I will close this, as I wrote my own driver based on this library, you can find it here, if you also want to control WS2816 Neopixels.
https://github.com/Jan-OleGiebel/LoRaWAN-LED-Christmastree

closed 2024-11-10 by Jan-OleGiebel

honnet, 2024-12-29:
Forked! (will try it on STM32C0)

For those looking for the driver only: \
https://github.com/Jan-OleGiebel/WS2816_Driver

Thank you for sharing it!

mcer12, 2025-01-27:
@Jan-OleGiebel **Thank you so much for this**, it's working well on STM32F030. I tried to port this to AVR (atmega328) but no dice so far, I'm just not good at this low level stuff at all :D I have some more STM32 so I'll just use those.