adafruit/Adafruit_NeoPixel #414: number of LEDs lead to stack overflow

roma08, 2024-12-02:
- Arduino board:  **ESP32-C3 or ESP32 WROOM 32**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **2.3.3**
- Library Adafruit NeoPixel Version 1.12.3
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
Hello team
I'm a beginner but I have some experience.
I've come across an error that I need your help with. Thank you for that.
If I load the sample from the website: https://github.com/adafruit/Adafruit_NeoPixel into my ESP32-C3, the application works perfectly. However, if I set the number of LEDs to <78, the ESP32-C3 has a problem and reports ***ERROR*** A stack overflow in task loopTask has been detected.
I have a project in which I want to control 144 - WS2812B LEDs. Can you help me make this project a success?
I have the same problem with ESP32 WROOM 32.  I'm attaching the code and the error message.
[AdafruitPixel_Case.txt](https://github.com/user-attachments/files/17983584/AdafruitPixel_Case.txt)
[Stakoverflow ESP32-C3.txt](https://github.com/user-attachments/files/17983536/Stakoverflow.ESP32-C3.txt)

Could you help me make this project a success?
Thanks Roland

thorathome, 2024-12-16:
I appear to be having the same problem. This is new behavior. 
Win 11
Arduino IDE 2.3.4
ESP32 WROOM-DA Module (Default 4MB with spiffs)
Adafruit NeoPixel Pebble 200-pixel seeds string (or nothing attached to the ESP32)

I get
```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
```

When using more than 74 pixels
```
Adafruit_NeoPixel pixelPanel ( 75, 25, NEO_GRB );
 ... 
  pinMode( 25, OUTPUT );        // sets the NeoPixel digital pin as output
  pixelPanel.begin();           // INITIALIZE NeoPixel pixelPanel object (REQUIRED)
  pixelPanel.show();            // Turn OFF all pixels ASAP

// Failure starts here //
```
All thoughts welcome. I cannot program my brand new Pebble string for now.  

Thank you.

ml-stu, 2024-12-30:
Same problem here. It works with version 1.11.0 (so downgrade the lib may help)
Starting from 1.12.0 I can not use more than 74 pixels in a string

thorathome, 2024-12-30:
Yow! You're a hero.  

Thank you for discovering this. Yes, downgrading to Adafruit_NeoPixel.h v 1.11.0 worked. 
Working with Adafruit's 200 pixel Pebble strings. Thanks.

ml-stu, 2024-12-31:
There are already some issues regarding this problem:
#402 
https://github.com/kitesurfer1404/WS2812FX/issues/368

dhalbert, 2025-01-12:
Closing as a duplicate in favor of #402

closed 2025-01-21 by dhalbert

thorathome, 2025-01-21:
**Nice work,** Adafruit team. I just tested Adafruit_NeoPixel v 1.12.4 with 200 pixels with no problems. 
Adafruit ESP32-S2 Feather
Espressif ESP32 board lib v 3.1.1
Arduino IDE on Win 11

**Thank you, as always, for maintaining these FREE libraries.** This is one of many reasons why I buy from Adafruit, even when available elsewhere. 

I purchased Adafruit's 200 pixel Pebble strings and was unable to use them until now.  

**Much much much appreciated.**

dhalbert, 2025-01-21:
Thank you to @teknynja and the testers and kibitzers.