adafruit/Adafruit_NeoPixel #421: RP2040 -  Changing setup parameters after initialization - Locks up PIO?

Chris-Beaty, 2025-02-05:
Hello,
The type of LED strand being used is not known at compile time.

If the init is changed after initialization (Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800); ) the data pin no longer transmits information.  ( confirmed using an oscilloscope )

                    /*  \\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\ 
                    //  \\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\ */
                    void Manage_NeoPixel_Type()
                      {
                          if(Neo_Pixel_Type != holdingRegs[Neo_Pixel_Type_Address] )
                            {
                              Neo_Pixel_Type = holdingRegs[Neo_Pixel_Type_Address]    ;     // Reset the variable to the new data

                              if (Neo_Pixel_Type == 0) {  Adafruit_NeoPixel strip(NUMPIXELS, Neo_Pixel_Pin, NEO_GRBW + NEO_KHZ400);   delay(10) ; strip.begin() ;  }
                              if (Neo_Pixel_Type == 1) {  Adafruit_NeoPixel strip(NUMPIXELS, Neo_Pixel_Pin, NEO_GRB  + NEO_KHZ400);   delay(10) ; strip.begin() ;  }
                              if (Neo_Pixel_Type == 2) {  Adafruit_NeoPixel strip(NUMPIXELS, Neo_Pixel_Pin, NEO_GRBW + NEO_KHZ800);   delay(10) ; strip.begin() ;  }
                              if (Neo_Pixel_Type == 3) {  Adafruit_NeoPixel strip(NUMPIXELS, Neo_Pixel_Pin, NEO_GRB  + NEO_KHZ800);   delay(10) ; strip.begin() ;  }

                              display.setCursor(0,45);
                              display.print("Neo_Pixel_Type ");  display.print(Neo_Pixel_Type); 
                              display.display();

                            }
                      }

Nianzu, 2025-05-08:
I believe I'm also running into this issue.

ladyada, 2025-05-11:
what if you create 4 global neopixel objects so you aren't put objects on the stack. then just begin() the one you want

closed 2025-05-28 by ladyada