adafruit/Adafruit_NeoPixel #429: ESP32-S3: show() will crash when used with WiFi

samleemc, 2025-04-10:
This is a test script I used. There are a few issues
1. The neopixel LED was not lid even through it is set to turn on in the setup()
2. When user clicked the "Turn on" button (or any button really) on the webpage, not only the neopixel will not turn on at all, but the board will also crash and reset. The terminal will show the print from this line Serial.println("show() led"); and then will see it printing the line from the setup() function again.

Hence, it seem like show() have issue with the wifi function in the ESP32-S3 board.

Board used : Adafruit ESP32-S3 Feather 8MB no PSRAM.
Adafruit NeoPixel : 1.12.5
 
```
#include <WiFi.h>
#include <WebServer.h>
#include <Adafruit_NeoPixel.h>

#define LED_PIN 33
#define NUM_PIXELS 1

Adafruit_NeoPixel pixels(NUM_PIXELS, LED_PIN, NEO_GRB + NEO_KHZ800);
WebServer server(80);

const char* ssid = "SSID";
const char* password = "12345678";

unsigned long lastBlink = 0;
bool blink = false;
bool ledOn = false;
uint32_t currentColor = pixels.Color(150, 150, 150);  // default white

// Flag to track WiFi connection status
bool wifiConnected = false;

void handleRoot() {
  String html = R"rawliteral(
    <!DOCTYPE html>
    <html>
    <head>
      <title>NeoPixel Control</title>
    </head>
    <body style="text-align:center;font-family:sans-serif;">
      <h2>NeoPixel LED Control</h2>
      <input type="color" id="colorPicker" value="#ffffff">
      <br><br>
      <button onclick="setMode('on')">Turn On</button>
      <button onclick="setMode('off')">Turn Off</button>
      <button onclick="setMode('blink')">Blink</button>
      <script>
        const colorPicker = document.getElementById("colorPicker");
        function setMode(mode) {
          const color = colorPicker.value.substring(1);
          fetch(`/set?mode=${mode}&color=${color}`);
        }
      </script>
    </body>
    </html>
  )rawliteral";

  server.send(200, "text/html", html);
}

void handleSet() {
  Serial.println("Received set command");
  Serial.print("API response OK");
  server.send(200, "text/plain", "OK");

  String mode = server.hasArg("mode") ? server.arg("mode") : "";
  Serial.print("mode = ");
  Serial.println(mode);

  String colorStr = server.hasArg("color") ? server.arg("color") : "ffffff";
  Serial.print("colorStr = ");
  Serial.println(colorStr);

  long r = strtol(colorStr.substring(0, 2).c_str(), NULL, 16);
  long g = strtol(colorStr.substring(2, 4).c_str(), NULL, 16);
  long b = strtol(colorStr.substring(4, 6).c_str(), NULL, 16);
  currentColor = pixels.Color(r, g, b);
  Serial.print("r = ");
  Serial.print®;

  Serial.print(" , g = ");
  Serial.print(g);
  Serial.print(" , b = ");
  Serial.println(b);

  if (mode == "on") {
    Serial.println("set blink and ledOn");
    blink = false;
    ledOn = true;
    Serial.println("set led color");
    pixels.setPixelColor(0, pixels.Color(150,0,0));
    Serial.println("show() led");
    pixels.show();
    Serial.println("after show() led");
  } else if (mode == "off") {
    blink = false;
    ledOn = false;
    pixels.clear();
    pixels.show();
  } else if (mode == "blink") {
    blink = true;
    ledOn = false;
    pixels.clear();
    pixels.show();
  }
}

void setup() {
  Serial.begin(9600);
  delay(1000);
  Serial.println("Starting up...");

  // Print reason for reset
  esp_reset_reason_t resetReason = esp_reset_reason();
  Serial.printf("Reset reason: %d\n", resetReason);

  Serial.println("Lightup White LED");
  delay(500);
  pixels.begin();
  pixels.setPixelColor(0, pixels.Color(150,150,150));
  pixels.show();

  // Connect to WiFi only once in setup()
  Serial.print("Connecting to WiFi");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected!");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());

  wifiConnected = true;  // Set the flag to true once connected

  server.on("/", handleRoot);
  server.on("/set", handleSet);
  server.begin();
}

void loop() {
  // Check if WiFi is connected before handling requests
  if (wifiConnected) {
    server.handleClient();
  }

  // Print available heap memory for diagnostics
  if (ESP.getFreeHeap() < 200000) {
    Serial.print("Free heap: ");
    Serial.println(ESP.getFreeHeap());
  }

  if (blink && millis() - lastBlink >= 1000) {
    static bool isOn = false;
    isOn = !isOn;
    if (isOn) {
      pixels.setPixelColor(0, currentColor);
    } else {
      pixels.clear();
    }
    pixels.show();
    lastBlink = millis();
  }
}
```

Nitroxid, 2025-04-10:
I have a similar issue with a QT Py ESP32-S3 (4MB/2MB)

I use the esp_now wifi functions to connect two MCUs. I wanted to use the neopixel as a "heartbeat" indicator, because I wanted some sort of response when the QT Py is not connected to a serial port.

I then ran into the same problem, that the QT Py was crashing every time pixels.show() was called.

keriszafir, 2025-04-18:
Works with esp32 v 3.1.3; broken with 3.2.0.