adafruit/Adafruit_NeoPixel #432: RMT error on ESP32-S3 since 1.12.4

Mark-Roly, 2025-04-22:
Hi.

Since updating the library to 1.12.4 the following error appears when first trying to call pixel.show();

`E (1027) rmt: rmt_new_tx_channel(269): not able to power down in light sleep`

This also occurs when using 1.12.5 (but does not appear when using 1.12.3).

Tested and replicated on multiple WEMOS S3 Mini units (ESP32-S3 based dev board) using the onboard WS28112B-087 (connected to IO47).

Issue replicated using:
- Adafruit_NeoPixel 1.12.4 and 1.12.5
- Arduino IDE 2.3.3 and 2.3.6
- ESP32 board 3.2.0
- Windows 11 Professional

**Sample sketch to replicate:**
```
#include <Adafruit_NeoPixel.h>
#define neopix_pin 47
#define NUMPIXELS 1
Adafruit_NeoPixel pixel = Adafruit_NeoPixel(NUMPIXELS, neopix_pin, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Adafruit_NeoPixel RMT issue tester");
  delay(1000);
  Serial.println("begin");
  pixel.begin();
  delay(1000);
  Serial.println("setPixelColor");
  pixel.setPixelColor(0, pixel.Color(0,0,25));
  delay(1000);
  Serial.println("show");
  pixel.show();
  Serial.println("Ready.");
}

void loop() {
  delay(10);
}
```

**Serial output from above sketch:**
```
Adafruit_NeoPixel RMT issue tester
begin
setPixelColor
show
E (3005) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
Ready.
```

nhbn, 2025-04-30:
Had the issue when updating existing ESP32-WROOM-32 - with in the meantime updated neopixel libraries.

In my case it lead to a reboot loop. After following MQTT functions had been called.

Reverting back to 1.12.3 of the neopixel library resolved the issue.

Pr77Pr77, 2025-05-01:
I had the same problem. Thank you for the workaround (downgrading to 1.12.3). A fix would be nice though.

ladyada, 2025-05-02:
@dhalbert do you have an idea (since i think ya merged the RMT fix)

pivoine31, 2025-05-02:
Sorry for making a dupe with #433 
If it can help : 
- in my case Neopixel lib 1.12.5 works correctly as far as  I don't reference the BME 680 library
- reverting to lib 1.12.3 permits to use both libraries flawlessly

Mark-Roly, 2025-05-17:
This issue is still present in 1.13.0

acefou, 2025-05-22:
hello, resolv with 1.14.0 for me (Lolin ESP32S3-MiniPro)

GryazniyNosok, 2025-05-22:
Had the same problem. Switching to 1.11.0 helped.

Mark-Roly, 2025-05-22:
> hello, resolv with 1.14.0 for me (Lolin ESP32S3-MiniPro)

I'm also seeing the same. Issue is resolved for me also in 1.14.0 using Wemos S3 Mini.
This ticket can now be closed.

Mark-Roly, 2025-05-23:
@dhalbert  Just updated my last comment - it seems that the error is still presenting in 1.14.0

TW-EK, 2025-05-27:
I can confirm the same issue happens with ESP32-C3 and ESP32-C6.   Using 1.12.3 works as a work around