adafruit/Adafruit_NeoPixel #433: Apparent incompatibility between Adafruit BME680 and Neopixel libraries

pivoine31, 2025-05-01:
When using simultaneously both libraries, at run time, calling the Neopixel library api (setPixelColor and show) leads to the following message (and strip don't emit any light)
rmt_new_tx_channel(269): not able to power down in light sleep
Using a Wemos D1 mini ESP32 ant tried with various pins for LED
(In the same situation, using FastLED don't produce the error and LEDs light up)

pivoine31, 2025-05-01:
Minimal code section to produce the problem (with latest libraries)

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include "Adafruit_BME680.h"

#define BME_SCK  18
#define BME_MISO 19
#define BME_MOSI 23
#define BME_CS    5

Adafruit_BME680 bme(BME_CS); // hardware SPI
//Adafruit_BME680 bme(BME_CS, BME_MOSI, BME_MISO, BME_SCK);

#include <Adafruit_NeoPixel.h>

#define LED_NUM 4
#define LED_PIN 16

Adafruit_NeoPixel pixels(LED_NUM, LED_PIN, NEO_GRB + NEO_KHZ800);

void set_led ( int rst ) {
  static int col[4] = { 0x00FF0000, 0x0000FF00, 0x000000FF, 0x007F7F7F };
  int i;
  for (i=0 ; i<LED_NUM ; i++)
    pixels.setPixelColor(i, col[i]);
  pixels.show();
}

void setup() {
  Serial.begin(115200);
  Serial.println("BME680 async test");
  set_led(1);
  Serial.println("LED setting done");
}

void loop() { }

ladyada, 2025-05-02:
dupe of #432

closed 2025-05-02 by ladyada