adafruit/Adafruit_NeoPixel #434: Support for RP2350B (GPIO 32-47)

sfe-SparkFro, 2025-05-07:
The RP2350B has 48 GPIO pins, compared to the 30 GPIO pins of the RP2350B and RP2040. Attempting to control LEDs on the higher GPIO pins (specifically, GPIO32-47) does not work.

It appears the [PIO driver](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/rp2040_pio.h) does not correctly set the PIO base if needed (required for GPIO 32-47, each PIO block can only access 32 GPIO). Suggest looking [here](https://github.com/peterharperuk/pico-examples/commit/7befc8d851830fc87f14630d9766cac0f8e8989b) for changes needed to get the higher GPIO working.

ladyada, 2025-05-11:
@sfe-SparkFro thanks, please submit a PR!

sfe-SparkFro, 2025-05-12:
Sorry, I do not have time to do any development work on this. I'm just reporting the problem since [one of our customers had this problem](https://xrp.discourse.group/t/driving-the-rgb-led-in-arduino/730), so just passing on the message.

closed 2025-05-28 by ladyada

sfe-SparkFro, 2025-05-28:
Seems to work great, thank you @ladyada!