adafruit/Adafruit_NeoPixel #437: Compilation error for Arduino Nano RP2040 Connect

humzakh, 2025-06-01:
Hi, I'm getting a compilation error using Adafruit_NeoPixel 1.15.1. Any help would be greatly appreciated.

```
/Users/hkhan/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_Neopixel_RP2.cpp: In member function 'bool Adafruit_NeoPixel::rp2040claimPIO()':
/Users/hkhan/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_Neopixel_RP2.cpp:9:9: error: 'pio_claim_free_sm_and_add_program_for_gpio_range' was not declared in this scope
   if (! pio_claim_free_sm_and_add_program_for_gpio_range(&ws2812_program,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hkhan/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_Neopixel_RP2.cpp: In member function 'void Adafruit_NeoPixel::rp2040releasePIO()':
/Users/hkhan/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_Neopixel_RP2.cpp:35:3: error: 'pio_remove_program_and_unclaim_sm' was not declared in this scope
   pio_remove_program_and_unclaim_sm(&ws2812_program, pio, pio_sm,  pio_program_offset);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hkhan/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_Neopixel_RP2.cpp:35:3: note: suggested alternative: 'pio_remove_program'
   pio_remove_program_and_unclaim_sm(&ws2812_program, pio, pio_sm,  pio_program_offset);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   pio_remove_program
exit status 1

Compilation error: exit status 1
```

je1rav, 2025-06-27:
The same error occurred on "Arduino Mbed OS RP2040 Boards by Arduino"

caternuson, 2025-06-27:
The Arduino Mbed OS Nano Board Package is using an older version of pico-sdk:
https://github.com/arduino/ArduinoCore-mbed/blob/main/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/generated/pico/version.h
that does not have the function being referenced:
https://github.com/arduino/ArduinoCore-mbed/blob/main/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_pio/include/hardware/pio.h

Here is a summary of the `-D`s added when compiling for a target of `Arduino Nano RP2040 Connect` using the `4.3.1` release of the BSP: 
```
-DARDUINO_ARCH_RP2040
-DARDUINO=10819
-DARDUINO_NANO_RP2040_CONNECT
-DARDUINO_ARCH_MBED_NANO
-DARDUINO_ARCH_MBED
-DARDUINO_LIBRARY_DISCOVERY_PHASE=0
```

Could update preproc logic to also check for `ARDUINO_ARCH_MBED` maybe?